espressif/esp-at #826: Disabling the default AP with SSID of esp_xxxxxx

fahirad, 2024-04-19:
### Answers checklist.

- [X] I have read the documentation [ESP-AT Programming Guide](https://docs.espressif.com/projects/esp-at/en/latest/) and the issue is not addressed there.
- [X] I have used the latest released firmware or have updated my ESP-AT branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

Hello all,

I see that when I flash my esp32c3 with esp-at project, the Wi-Fi driver gets initialized by default and both access point and station mode are started, resulting in a new network with a name ESP_xxxxxx where these x's represent the last 6 characters of the MAC address. 
How do I disable it? I want to be able to enable the driver itself and the custom ap network in certain cases, so I don't need this default behaviour. 
I am familiar with the command AT+CWINIT=0 which will disable it, but how do I save this setting so that on reset it doesn't get started again?

Thanks ahead for any clarification!

bschousek, 2024-04-22:
Have you confirmed that the AT+CWINIT=0 is not persistent through reset? I seem to recall that issuing the command once writes it into some non-volatile memory that only gets changed the next time that you flash the firmware.

ustccw, 2024-05-17:
@fahirad @bschousek you can send `AT+CWMODE=0` to make it.

closed 2025-03-17 by ustccw