espressif/esp-at #827: No way to get HTTP Error code

Rsaliu, 2024-05-02:
### Answers checklist.

- [X] I have read the documentation [ESP-AT Programming Guide](https://docs.espressif.com/projects/esp-at/en/latest/) and the issue is not addressed there.
- [X] I have used the latest released firmware or have updated my ESP-AT branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

I am working with esp-at. 

I can successfully send a post request using:

AT+HTTPCPOST

After sending the post request, I got the following output with SEND OK which means that it is successful.

[image]

 However, I can't get the error code. According to the document, it says to AT+HTTPCLIENT to get the error code but that always results in an error.

[image]

 
 Kindly help on how to get HTTP error code after post request.

ustccw, 2024-05-17:
@Rsaliu i'm sorry that `AT+HTTPCPOST` command does not provide a HTTP errrno after post request. it always returns the `ERROR` if `errno > 400`.  
We will consider whether to add an HTTP error code in this case. As a workaround, you can catch the errno in `esp_http_client_get_status_code()` API, and send it by `esp_at_port_write_data()`.

JavierReyes945, 2024-10-31:
We are also having this issue, even in the current master branch. It is really unconvenient to not have a way to check the return code of the HTTP transaction.

> As a workaround, you can catch the errno in esp_http_client_get_status_code() API, and send it by esp_at_port_write_data()

Not sure what is the meaning of this, maybe you got confused of repository? This is ESP AT Firmware, so we only have AT commands, not C functions on ESP-IDF framework.

tmpfs, 2025-06-22:
I just want to add that this is super annoying. We are using ESP-AT (not ESP-IDF) so can't use the C functions mentioned.

I have found a workaround though, if the server replies with the code in the body it can be parsed from a URC, for example I see this URC in the logs:

```
+HTTPCPOST:3,403\r\n
```

So now we have to parse the response code from the response body, sigh.

Please ESP-AT authors add HTTP response status codes to all ERROR responses, I mean it's even in the [docs](https://docs.espressif.com/projects/esp-at/en/latest/esp32/AT_Command_Set/HTTP_AT_Commands.html#cmd-httperrcode).

closed 2025-06-27 by igrr

ustccw, 2025-06-30:
It is supported on latest master, see [this](https://docs.espressif.com/projects/esp-at/en/latest/esp32/AT_Command_Set/HTTP_AT_Commands.html#http-at-error-codes) for more details.