espressif/esp-at #828: Packet loss in TCP

venkateshdhara, 2024-05-05:
### Answers checklist.

- [X] I have read the documentation [ESP-AT Programming Guide](https://docs.espressif.com/projects/esp-at/en/latest/) and the issue is not addressed there.
- [X] I have used the latest released firmware or have updated my ESP-AT branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

I am using ESP32-C3 and experiencing packet loss in reception in TCP and transmit is always good, following methods are debugged
system is setup in such a way ESP32 is connected to stm32l4 MCU and exchanging data every 5seconds with raspberry pi
1.Ran TCPDUMP service in RPI there is no tcp retransmission request observed.
2.ESP32 using in UART 115200 tried changing baud rate 230400.
3.Tried CIPRECTYPE=1. passive mode
4.probed logic analyzer on UART and confirmed data was not received on UART.
5.KEEPALIVE value 3 sec
6.used quick configuration to increase throughput as suggested in FAQ page 
https://docs.espressif.com/projects/esp-at/en/latest/esp32/Compile_and_Develop/How_to_optimize_throughput.html#simple-quick-configuration

ustccw, 2024-05-17:
You mentioned there is no tcp retransmission, and i assume your `STM32L4` is good, the packet loss should happen on the UART  transmission.  
Could you please enable the uart hardware flow control by adding [CTS/RTS line](https://docs.espressif.com/projects/esp-at/en/latest/esp32c3/Get_Started/Hardware_connection.html#id3) here, and enables it by sending [AT+UART_CUR=230400,8,1,0,3](https://docs.espressif.com/projects/esp-at/en/latest/esp32c3/AT_Command_Set/Basic_AT_Commands.html#at-uart-cur-current-uart-configuration-not-saved-in-flash)

venkateshdhara, 2024-05-30:
@ustccw i have done this experiment by using qcom terminal sending message to hurceles software(tcp server) and receiving echo with 5sec interval still i see packet loss, i have done this experiment before seeing your comment but i did this at 115200 with hardware flow control CTS/RTS

ustccw, 2024-07-01:
@venkateshdhara If you have already enabled hardware flow control and other conditions outside are right, this should be a bug.

Could you please use latest master branch and enable more logs to see what happened:
```
python build.py menuconfig -> Component config -> Log output -> Default log verbosity -> Debug
python build.py menuconfig -> Component config -> AT -> Enable ESP-AT Debug
python build.py menuconfig -> Component config -> AT -> Enable ESP-AT Debug -> Enable Network Debug
(Disable BT to reduce bin size) python build.py menuconfig -> Component config -> Bluetooth -> Deselect
```

The AT log port output should give more logs.

venkateshdhara, 2024-07-10:
i will post soon

venkateshdhara, 2024-07-10:
i will post soon

venkateshdhara, 2024-12-02:
Hi I have conducted experiments with raspberry pi 3 both  on ethernet and wifi communicating tcp with esp32 at seems something to be tweaked at antenna no issue in firmware, I hope issue can be closed

closed 2024-12-02 by venkateshdhara