espressif/esp-at #829: AT 固件 BLE 连接断开原因

ceerqingtingml, 2024-05-09:
### Answers checklist.

- [X] I have read the documentation [ESP-AT Programming Guide](https://docs.espressif.com/projects/esp-at/en/latest/) and the issue is not addressed there.
- [X] I have used the latest released firmware or have updated my ESP-AT branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

AT 固件如何获取 BLE 连接断开的原因？

ustccw, 2024-05-17:
@Luxin812 PTAL

Luxin812, 2024-05-28:
目前 AT 官方固件不能获取到 BLE 连接断开的原因，你可以在 menuconfig 中将 HCI 日志打开重新编译固件，menuconfig 更改如下：
python build.py menuconfig -> Component config -> Log output -> Default log verbosity -> Debug
python build.py menuconfig -> Component config -> Bluetooth -> Bluedroid Options -> BT DEBUG LOG LEVEL -> HCI layer -> Debug
另外在编译固件时不要将固件设置为 slience 模式
如何编译 AT 固件请参考：https://docs.espressif.com/projects/esp-at/zh_CN/latest/esp32/Compile_and_Develop/How_to_clone_project_and_compile_it.html#esp-at 或者是：https://docs.espressif.com/projects/esp-at/zh_CN/latest/esp32/Compile_and_Develop/How_to_build_project_with_web_page.html#esp-at
BLE 的错误码可以直接看 BLE 5.0 协议里面的 ERROR CODE 部分：
[image]

Luxin812, 2024-08-30:
目前最新 master 分支上的 AT 固件（因为 ESP32 系列是 silence mode，所以如果您使用的是 ESP32 AT 固件，需要自行在[本地编译](https://docs.espressif.com/projects/esp-at/zh_CN/latest/esp32/Compile_and_Develop/How_to_clone_project_and_compile_it.html#esp-at) AT 固件，禁用 silence mode），可以在 AT 日志端口获取到 BLE 断开的原因。

closed 2025-03-17 by ustccw