espressif/esp-at #832: How does modem wake up from light sleep mode.

EdwardRyu, 2024-05-13:
### Answers checklist.

- [X] I have read the documentation [ESP-AT Programming Guide](https://docs.espressif.com/projects/esp-at/en/latest/) and the issue is not addressed there.
- [X] I have used the latest released firmware or have updated my ESP-AT branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

Hello,

Our product is using below version.

**HW :**			
[ESP32-WROVER-E]

**SW**
IDF-v4.4.4
ADF-V2.5
AT-v2.4.0.0

I have a couple of questions about sleep mode as follows.

1. Can device wake up from light sleep mode by receiving AT commands from application processor?
2. Can device wake up from light sleep mode when receiving packets from any of Ethernet or Wi-Fi?

Thanks,
Edward

ustccw, 2024-05-17:
@EdwardRyu 
1. No, this is the wakeup mode by uart. esp-at doesn't support it now. please refer to [AT+SLEEPWKCFG](https://docs.espressif.com/projects/esp-at/en/latest/esp32/AT_Command_Set/Basic_AT_Commands.html#cmd-wkcfg) for more details.
2. Yes, device will auto wake up from light sleep, and do [ESP-AT Message Reports](https://docs.espressif.com/projects/esp-at/en/latest/esp32/AT_Command_Set/index.html#id4), MCU can receive network data. please refer to [Sleep AT Examples](https://docs.espressif.com/projects/esp-at/en/latest/esp32/AT_Command_Examples/sleep_at_examples.html) for more details.

EdwardRyu, 2024-05-17:
Thank you for your answer.

Let me ask another question.
We are using ESP32 as a modem, and we have another controller MCU for applications.
We want to control sleep mode of ESP32 from the application processor.

Could you guide how to get ESP32 into sleep mode and wake it up?
For example, if I have a look into [AT+SLEEPWKCFG](https://docs.espressif.com/projects/esp-at/en/latest/esp32/AT_Command_Set/Basic_AT_Commands.html#cmd-wkcfg), I could see ESP-AT supports wake up using GPIO.

However, we are using GPIO12 for other purpose. And we have no other GPIO available at the moment.

My questions are ;
Q1) Do you have any good idea on this circumstance?
Q2) Can we use other GPIO rather than GPIO12 by configuring through AT+SLEEPWKCFG?

Thank you,
Edward

ustccw, 2024-05-20:
@EdwardRyu Of course, you can use any unused GPIO with input capability as the wakeup GPIO. maybe you can refer to your module datasheet for more details.

EdwardRyu, 2024-05-27:
Hi,

I have issued following commands in order to put modem to light sleep mode.

- AT+CWMODE=1
- AT+CWJAP=,,,,,3
- AT+SLEEP=2

Right after issuing those at commands, I could see some drops of current consumptions by modem from 120 mA to 48 mA.

In this sleep mode, 
- I am able to send and receive AT commands properly.
- I could receive packets from internet as well.
- Current consumption is still less than before entering light sleep mode, but it is fluctuating.

Questions are :

Q1) I am not sure whether if modem entered in light sleep mode or not. Could you guide how to check if modem is in light sleep mode or not?
As you can see, below log snippets below, I couldn't see any sleep specific AT response from modem after AT+SLEEP=2.

Q2) What is the range of current consumption by modem in light sleep mode?
When I look into data sheet from below link it mentioned between 0.8 mA.
But the gap between our measure and the data in data sheet is far different.
Could you tell me more about this?

Q3) In this mode, when I send AT command from MCU to modem, I could get response from modem as usual.
I wonder if modem will be getting into sleep mode again automatically or not after getting AT response from modem.
Or do we have to issue any AT+SLEEP command to modem again?

[2024-05-24 16:33:28.735] [0] [ATT][MODM] AT+CWMODE=1
[2024-05-24 16:33:28.735] 
[2024-05-24 16:33:28.766] [2] [ATR][19,0x200004fc]
[2024-05-24 16:33:28.766]  AT+CWMODE=1
[2024-05-24 16:33:28.766] 
[2024-05-24 16:33:28.766] OK

[2024-05-24 16:33:34.032] [2] [ATT][MODM] AT+CWJAP=,,,,,3
[2024-05-24 16:33:34.032] 
[2024-05-24 16:33:34.079] [4] [ATR][55,0x200004fc]
[2024-05-24 16:33:34.079]  AT+CWJAP=,,,,,3
[2024-05-24 16:33:34.079] WIFI DISCONNECT
[2024-05-24 16:33:34.079] +MQTTDISCONNECTED:0
[2024-05-24 16:33:34.110] [8] [ATR][16,0x200004fc]
[2024-05-24 16:33:34.110]  WIFI CONNECTED
[2024-05-24 16:33:35.128] [7] [ATR][19,0x200004fc]
[2024-05-24 16:33:35.128]  
[2024-05-24 16:33:35.128] OK
[2024-05-24 16:33:35.128] WIFI GOT IP

[2024-05-24 16:35:35.220] [0] [ATT][MODM] AT+SLEEP=2
[2024-05-24 16:35:35.220] [2] [ATR][18,0x200004fc]
[2024-05-24 16:35:35.236]  AT+SLEEP=2
[2024-05-24 16:35:35.236] 
[2024-05-24 16:35:35.236] OK

Thanks,
Edward

ustccw, 2024-07-01:
@EdwardRyu May i know which firmware you use? and would you mind trying to compile your firmware based on [master branch](https://github.com/espressif/esp-at/tree/master).

In generally:
### Q1
If the average current of your module is less than 5mA, it should have entered in light-sleep. During light-sleep, AT is not able to respond the AT commands normally. 
Have you read the [this document](https://docs.espressif.com/projects/esp-at/en/latest/esp32/AT_Command_Examples/sleep_at_examples.html#measurement-method) before, I am not sure how you measure the current, are you sure using the module, rather than a development board?

### Q2
I don't think your module have entered light-sleep mode.

### Q3
We did not provide a guarantee for a range of average current values. But as a test, during the light-sleep, the average current should be less than 5mA. The data about auto light-sleep, you can refer to [this doc](https://docs.espressif.com/projects/esp-idf/zh_CN/latest/esp32/api-guides/low-power-mode.html#id8), see the `auto light Sleep` column.

> `0.8mA` is the chip current, not the module current. Moreover, esp-at uses the auto light-sleep to keep the Wi-Fi connection and network connection, rather than the normal light-sleep that datasheet mentioned.

### Q4
esp-at does not respond to any AT commands during light-sleep. I don't think your module have entered light-sleep mode.
If the Wi-Fi module enters light-sleep, there is no need to send the `AT+SLEEP` command again. The device will automatically sleep and wake up based on the wake-up config ([AT+SLEEPWKCFG](https://docs.espressif.com/projects/esp-at/zh_CN/latest/esp32/AT_Command_Set/Basic_AT_Commands.html#basic-at)) and network status.

### Q5
You can use any unused GPIO with input capability as the wakeup GPIO.
- `AT+SLEEPWKCFG=2,28,0` returned ERROR, because of GPIO28 is not a available GPIO to users.
- `AT+SLEEPWKCFG=2,12,0` returned `GPIO isr service already installed`, it seems that you have already sent this command before. Could you please provide a complete log if issue still exists.

~

EdwardRyu, 2024-07-04:
Hello,

Thanks for your feedback.

Our test environment is like attached snap shot.
We cut the board and connected two points as described as your post.
[image]

When I test the original ESP-AT V2.4 over our development board, I could see light sleep current is less than 2mA.
After enabling ETHERNET feature with below option, however, then CPU cannot go into light sleep mode.
As you can see, it is above 25 mA although 'AT+SLEEP=2' returned OK.
> CONFIG_AT_ETHERNET_SUPPORT=y
> # CONFIG_PHY_IP101 is not set
> CONFIG_PHY_RTL8201=y
> # CONFIG_PHY_DP83848 is not set
> # CONFIG_PHY_LAN8720 is not set

Could you please check and let me know what the problem is here?

Thanks,
Edward

ustccw, 2024-10-10:
esp-at doesn't test power consumption when ethernet enabled, maybe you can try to optimise it in esp-idf.

ustccw, 2024-10-10:
@EdwardRyu We sincerely apologize that [esp-idf](https://github.com/espressif/esp-idf) has not yet optimized the power consumption for Ethernet usage. This means that when Ethernet is enabled, the module may not be able to enter light-sleep, resulting in relatively higher power consumption.

If this feature is essential for your product, we kindly suggest submitting detailed requirements to [esp-idf issue](https://github.com/espressif/esp-idf/issues/new?assignees=&labels=Type%3A+Feature+Request&projects=&template=03_feature_request.yml).

closed 2025-03-17 by ustccw