espressif/esp-at #842: ESP32 C3 MINI Build Error

Aravind0812, 2024-05-30:
### Answers checklist.

- [X] I have read the documentation [ESP-AT Programming Guide](https://docs.espressif.com/projects/esp-at/en/latest/) and the issue is not addressed there.
- [X] I have used the latest released firmware or have updated my ESP-AT branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

While building ESP32 C3 MINI after enabled AT+FS and Webserver when creating fatfs.bin getting invalid argument error. So i manual run that python without the argument "--wl_mode "safe" and generated fatfs.bin after and then i build it.

This is that error:
wl_fatfsgen.py: error: unrecognized arguments: --wl_mode safe
generating fatfs.bin: `C:\Espressif\python_env\idf5.2_py3.11_env\Scripts\python.exe C:/Users/username/esp/esp-at\esp-idf\components\fatfs\wl_fatfsgen.py C:/Users/username/esp/esp-at\components\fs_image --partition_size 102400 --output_file C:/Users/username/esp/esp-at/build/customized_partitions\fatfs.bin --sector_size 512 --long_name_support --sectors_per_cluster 1 --fat_type 0 --wl_mode "safe"`
Please check the log below.
```
[0/1] Re-running CMake...
-- silence:1, sdkconfig:C:/Users/username/esp/esp-at/module_config/module_esp32c3_default/sdkconfig_silence.defaults
-- partition_table_dir: C:/Users/username/esp/esp-at
-- Component directory C:/Users/username/esp/esp-at/components/fs_image does not contain a CMakeLists.txt file. No component will be added
-- ccache will be used for faster recompilation
-- Building ESP-IDF components for target esp32c3
NOTICE: Skipping optional dependency: espressif/cmake_utilities
Processing 6 dependencies:
[1/6] espressif/bootloader_support_plus (0.2.2)
[2/6] espressif/cmake_utilities (0.5.3)
[3/6] espressif/esp_websocket_client (1.2.3)
[4/6] espressif/mdns (1.3.1)
[5/6] espressif/xz (1.0.0)
[6/6] idf (5.0.6)
NOTICE: Skipping optional dependency: espressif/cmake_utilities
-- Project sdkconfig file C:/Users/username/esp/esp-at/sdkconfig
Loading defaults file C:/Users/username/esp/esp-at/module_config/module_esp32c3_default/sdkconfig_silence.defaults...
-- App "esp-at" version: v3.3.0.0-20-ga24d4fb4
-- Adding linker script C:/Users/username/esp/esp-at/build/esp-idf/esp_system/ld/memory.ld
-- Adding linker script C:/Users/username/esp/esp-at/esp-idf/components/esp_system/ld/esp32c3/sections.ld.in
-- Adding linker script C:/Users/username/esp/esp-at/esp-idf/components/esp_rom/esp32c3/ld/esp32c3.rom.ld
-- Adding linker script C:/Users/username/esp/esp-at/esp-idf/components/esp_rom/esp32c3/ld/esp32c3.rom.api.ld
-- Adding linker script C:/Users/username/esp/esp-at/esp-idf/components/esp_rom/esp32c3/ld/esp32c3.rom.libgcc.ld
-- Adding linker script C:/Users/username/esp/esp-at/esp-idf/components/esp_rom/esp32c3/ld/esp32c3.rom.newlib.ld
-- Adding linker script C:/Users/username/esp/esp-at/esp-idf/components/esp_rom/esp32c3/ld/esp32c3.rom.version.ld
-- Adding linker script C:/Users/username/esp/esp-at/esp-idf/components/esp_rom/esp32c3/ld/esp32c3.rom.eco3.ld
-- Adding linker script C:/Users/username/esp/esp-at/esp-idf/components/soc/esp32c3/ld/esp32c3.peripherals.ld
-- BOOTLOADER_SUPPORT_PLUS: 0.2.2
-- Set custom partition table: C:/Users/username/esp/esp-at/module_config/module_esp32c3_default/at_customize.csv
-- Components: app_trace app_update at bootloader bootloader_support bt cmock console customized_partitions cxx driver efuse esp-tls esp_adc esp_app_format esp_common esp_eth esp_event esp_gdbstub esp_hid esp_http_client esp_http_server esp_https_ota esp_https_server esp_hw_support esp_lcd esp_local_ctrl esp_netif esp_partition esp_phy esp_pm esp_psram esp_ringbuf esp_rom esp_system esp_timer esp_wifi espcoredump espressif__bootloader_support_plus espressif__cmake_utilities espressif__esp_websocket_client espressif__mdns espressif__xz esptool_py fatfs freertos hal heap http_parser idf_test ieee802154 json log lwip main mbedtls mqtt newlib nvs_flash openthread partition_table protobuf-c protocomm pthread riscv sdmmc soc spi_flash spiffs tcp_transport ulp unity usb vfs wear_levelling wifi_provisioning wpa_supplicant
-- Component paths: C:/Users/username/esp/esp-at/esp-idf/components/app_trace C:/Users/username/esp/esp-at/esp-idf/components/app_update C:/Users/username/esp/esp-at/components/at C:/Users/username/esp/esp-at/esp-idf/components/bootloader C:/Users/username/esp/esp-at/esp-idf/components/bootloader_support C:/Users/username/esp/esp-at/esp-idf/components/bt C:/Users/username/esp/esp-at/esp-idf/components/cmock C:/Users/username/esp/esp-at/esp-idf/components/console C:/Users/username/esp/esp-at/components/customized_partitions C:/Users/username/esp/esp-at/esp-idf/components/cxx C:/Users/username/esp/esp-at/esp-idf/components/driver C:/Users/username/esp/esp-at/esp-idf/components/efuse C:/Users/username/esp/esp-at/esp-idf/components/esp-tls C:/Users/username/esp/esp-at/esp-idf/components/esp_adc C:/Users/username/esp/esp-at/esp-idf/components/esp_app_format C:/Users/username/esp/esp-at/esp-idf/components/esp_common C:/Users/username/esp/esp-at/esp-idf/components/esp_eth C:/Users/username/esp/esp-at/esp-idf/components/esp_event C:/Users/username/esp/esp-at/esp-idf/components/esp_gdbstub C:/Users/username/esp/esp-at/esp-idf/components/esp_hid C:/Users/username/esp/esp-at/esp-idf/components/esp_http_client C:/Users/username/esp/esp-at/esp-idf/components/esp_http_server C:/Users/username/esp/esp-at/esp-idf/components/esp_https_ota C:/Users/username/esp/esp-at/esp-idf/components/esp_https_server C:/Users/username/esp/esp-at/esp-idf/components/esp_hw_support C:/Users/username/esp/esp-at/esp-idf/components/esp_lcd C:/Users/username/esp/esp-at/esp-idf/components/esp_local_ctrl C:/Users/username/esp/esp-at/esp-idf/components/esp_netif C:/Users/username/esp/esp-at/esp-idf/components/esp_partition C:/Users/username/esp/esp-at/esp-idf/components/esp_phy C:/Users/username/esp/esp-at/esp-idf/components/esp_pm C:/Users/username/esp/esp-at/esp-idf/components/esp_psram C:/Users/username/esp/esp-at/esp-idf/components/esp_ringbuf C:/Users/username/esp/esp-at/esp-idf/components/esp_rom C:/Users/username/esp/esp-at/esp-idf/components/esp_system C:/Users/username/esp/esp-at/esp-idf/components/esp_timer C:/Users/username/esp/esp-at/esp-idf/components/esp_wifi C:/Users/username/esp/esp-at/esp-idf/components/espcoredump C:/Users/username/esp/esp-at/managed_components/espressif__bootloader_support_plus C:/Users/username/esp/esp-at/managed_components/espressif__cmake_utilities C:/Users/username/esp/esp-at/managed_components/espressif__esp_websocket_client C:/Users/username/esp/esp-at/managed_components/espressif__mdns C:/Users/username/esp/esp-at/managed_components/espressif__xz C:/Users/username/esp/esp-at/esp-idf/components/esptool_py C:/Users/username/esp/esp-at/esp-idf/components/fatfs C:/Users/username/esp/esp-at/esp-idf/components/freertos C:/Users/username/esp/esp-at/esp-idf/components/hal C:/Users/username/esp/esp-at/esp-idf/components/heap C:/Users/username/esp/esp-at/esp-idf/components/http_parser C:/Users/username/esp/esp-at/esp-idf/components/idf_test C:/Users/username/esp/esp-at/esp-idf/components/ieee802154 C:/Users/username/esp/esp-at/esp-idf/components/json C:/Users/username/esp/esp-at/esp-idf/components/log C:/Users/username/esp/esp-at/esp-idf/components/lwip C:/Users/username/esp/esp-at/main C:/Users/username/esp/esp-at/esp-idf/components/mbedtls C:/Users/username/esp/esp-at/esp-idf/components/mqtt C:/Users/username/esp/esp-at/esp-idf/components/newlib C:/Users/username/esp/esp-at/esp-idf/components/nvs_flash C:/Users/username/esp/esp-at/esp-idf/components/openthread C:/Users/username/esp/esp-at/esp-idf/components/partition_table C:/Users/username/esp/esp-at/esp-idf/components/protobuf-c C:/Users/username/esp/esp-at/esp-idf/components/protocomm C:/Users/username/esp/esp-at/esp-idf/components/pthread C:/Users/username/esp/esp-at/esp-idf/components/riscv C:/Users/username/esp/esp-at/esp-idf/components/sdmmc C:/Users/username/esp/esp-at/esp-idf/components/soc C:/Users/username/esp/esp-at/esp-idf/components/spi_flash C:/Users/username/esp/esp-at/esp-idf/components/spiffs C:/Users/username/esp/esp-at/esp-idf/components/tcp_transport C:/Users/username/esp/esp-at/esp-idf/components/ulp C:/Users/username/esp/esp-at/esp-idf/components/unity C:/Users/username/esp/esp-at/esp-idf/components/usb C:/Users/username/esp/esp-at/esp-idf/components/vfs C:/User [... cut ...]
  File "C:\Users\username\esp\esp-at\tools\esp_at_factory_bin_combine.py", line 58, in main
    esp_at_combine_bin(args.module_name.upper(), args.flash_mode.upper(), args.flash_size.upper(),
  File "C:\Users\username\esp\esp-at\tools\esp_at_factory_bin_combine.py", line 29, in esp_at_combine_bin
    with open(os.path.join(build_dir, bin_file), 'rb') as f:
         ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
FileNotFoundError: [Errno 2] No such file or directory: 'C:/Users/username/esp/esp-at/build\\customized_partitions/fatfs.bin'
ninja: build stopped: subcommand failed.
```

ustccw, 2024-06-28:
@Aravind0812 Hello, it seems that 
```
```

Aravind0812, 2024-07-11:
I am getting `fatfs_generation.patch does not need to be applied.`

**Full Log:**
```
` python build.py build
Platform name:ESP32C3   Module name:MINI-1      Silence:1
submodules check completed for updates.
esp_tls_parse_ecc_key.patch does not need to be applied.
fatfs_generation.patch does not need to be applied.
support_ext_partition.patch does not need to be applied.
patches check completed for updates.
Ready to set up environment variables..
PATH is C:\Espressif\tools\xtensa-esp-elf-gdb\12.1_20231023\xtensa-esp-elf-gdb\bin;C:\Espressif\tools\riscv32-esp-elf-gdb\12.1_20231023\riscv32-esp-elf-gdb\bin;C:\Espressif\tools\xtensa-esp-elf\esp-13.2.0_20230928\xtensa-esp-elf\bin;C:\Espressif\tools\riscv32-esp-elf\esp-13.2.0_20230928\riscv32-esp-elf\bin;C:\Espressif\tools\esp32ulp-elf\2.35_20220830\esp32ulp-elf\bin;C:\Espressif\tools\cmake\3.24.0\bin;C:\Espressif\tools\openocd-esp32\v0.12.0-esp32-20230921\openocd-esp32\bin;C:\Espressif\tools\ninja\1.11.1\;C:\Espressif\tools\idf-exe\1.0.3\;C:\Espressif\tools\ccache\4.8\ccache-4.8-windows-x86_64;C:\Espressif\tools\dfu-util\0.11\dfu-util-0.11-win64;C:\Espressif\frameworks\esp-idf-v5.2.1\tools;C:\Espressif\tools\idf-git\2.43.0\cmd;C:\Espressif\python_env\idf5.2_py3.11_env\Scripts;C:\Espressif;C:\WINDOWS\system32;C:\WINDOWS;C:\WINDOWS\System32\Wbem;C:\WINDOWS\System32\WindowsPowerShell\v1.0\;C:\WINDOWS\System32\OpenSSH\;C:\Program Files\PuTTY\;C:\Program Files (x86)\STMicroelectronics\STM32 ST-LINK Utility\ST-LINK Utility;C:\Program Files\Docker\Docker\resources\bin;C:\Users\aidea\AppData\Local\Microsoft\WindowsApps
IDF_PYTHON_ENV_PATH is C:\Espressif\python_env\idf5.2_py3.11_env
sys.platform is win32
Not using an unsupported version of tool xtensa-esp32-elf found in PATH: esp-13.2.0_20230928-13.2.0. To use it, run 'C:\Espressif\python_env\idf5.2_py3.11_env\Scripts\python.exe C:\Users\aravindsamy\esp\esp-at\esp-idf\tools\idf_tools.py export --prefer-system'
Not using an unsupported version of tool riscv32-esp-elf found in PATH: esp-13.2.0_20230928-13.2.0. To use it, run 'C:\Espressif\python_env\idf5.2_py3.11_env\Scripts\python.exe C:\Users\aravindsamy\esp\esp-at\esp-idf\tools\idf_tools.py export --prefer-system'
Not using an unsupported version of tool ninja found in PATH: 1.11.1. To use it, run 'C:\Espressif\python_env\idf5.2_py3.11_env\Scripts\python.exe C:\Users\aravindsamy\esp\esp-at\esp-idf\tools\idf_tools.py export --prefer-system'
Not using an unsupported version of tool ccache found in PATH: 4.8. To use it, run 'C:\Espressif\python_env\idf5.2_py3.11_env\Scripts\python.exe C:\Users\aravindsamy\esp\esp-at\esp-idf\tools\idf_tools.py export --prefer-system'
export str is ESP_ROM_ELF_DIR=C:\Espressif\tools\esp-rom-elfs\20220823\
ESP_IDF_VERSION=5.0
PATH=C:\Espressif\tools\xtensa-esp-elf-gdb\12.1_20231023\xtensa-esp-elf-gdb\bin;C:\Espressif\tools\riscv32-esp-elf-gdb\12.1_20231023\riscv32-esp-elf-gdb\bin;C:\Espressif\tools\xtensa-esp32-elf\esp-2022r1-11.2.0\xtensa-esp32-elf\bin;C:\Espressif\tools\riscv32-esp-elf\esp-2022r1-11.2.0\riscv32-esp-elf\bin;C:\Espressif\tools\esp32ulp-elf\2.35_20220830\esp32ulp-elf\bin;C:\Espressif\tools\cmake\3.24.0\bin;C:\Espressif\tools\openocd-esp32\v0.12.0-esp32-20230921\openocd-esp32\bin;C:\Espressif\tools\ninja\1.10.2\;C:\Espressif\tools\idf-exe\1.0.3\;C:\Espressif\tools\ccache\4.6.2\ccache-4.6.2-windows-x86_64;C:\Users\aravindsamy\esp\esp-at\esp-idf\tools;%PATH%
IDF_DEACTIVATE_FILE_PATH=C:\Users\aidea\AppData\Local\Temp\tmpvzq4yl_5idf_33096

PATH is C:\Espressif\tools\xtensa-esp-elf-gdb\12.1_20231023\xtensa-esp-elf-gdb\bin;C:\Espressif\tools\riscv32-esp-elf-gdb\12.1_20231023\riscv32-esp-elf-gdb\bin;C:\Espressif\tools\xtensa-esp32-elf\esp-2022r1-11.2.0\xtensa-esp32-elf\bin;C:\Espressif\tools\riscv32-esp-elf\esp-2022r1-11.2.0\riscv32-esp-elf\bin;C:\Espressif\tools\esp32ulp-elf\2.35_20220830\esp32ulp-elf\bin;C:\Espressif\tools\cmake\3.24.0\bin;C:\Espressif\tools\openocd-esp32\v0.12.0-esp32-20230921\openocd-esp32\bin;C:\Espressif\tools\ninja\1.10.2\;C:\Espressif\tools\idf-exe\1.0.3\;C:\Espressif\tools\ccache\4.6.2\ccache-4.6.2-windows-x86_64;C:\Users\aravindsamy\esp\esp-at\esp-idf\tools;%PATH%;C:\Espressif\tools\xtensa-esp-elf-gdb\12.1_20231023\xtensa-esp-elf-gdb\bin;C:\Espressif\tools\riscv32-esp-elf-gdb\12.1_20231023\riscv32-esp-elf-gdb\bin;C:\Espressif\tools\xtensa-esp-elf\esp-13.2.0_20230928\xtensa-esp-elf\bin;C:\Espressif\tools\riscv32-esp-elf\esp-13.2.0_20230928\riscv32-esp-elf\bin;C:\Espressif\tools\esp32ulp-elf\2.35_20220830\esp32ulp-elf\bin;C:\Espressif\tools\cmake\3.24.0\bin;C:\Espressif\tools\openocd-esp32\v0.12.0-esp32-20230921\openocd-esp32\bin;C:\Espressif\tools\ninja\1.11.1\;C:\Espressif\tools\idf-exe\1.0.3\;C:\Espressif\tools\ccache\4.8\ccache-4.8-windows-x86_64;C:\Espressif\tools\dfu-util\0.11\dfu-util-0.11-win64;C:\Espressif\frameworks\esp-idf-v5.2.1\tools;C:\Espressif\tools\idf-git\2.43.0\cmd;C:\Espressif\python_env\idf5.2_py3.11_env\Scripts;C:\Espressif;C:\WINDOWS\system32;C:\WINDOWS;C:\WINDOWS\System32\Wbem;C:\WINDOWS\System32\WindowsPowerShell\v1.0\;C:\WINDOWS\System32\OpenSSH\;C:\Program Files\PuTTY\;C:\Program Files (x86)\STMicroelectronics\STM32 ST-LINK Utility\ST-LINK Utility;C:\Program Files\Docker\Docker\resources\bin;C:\Users\aidea\AppData\Local\Microsoft\WindowsApps
IDF_PYTHON_ENV_PATH is C:\Espressif\python_env\idf5.2_py3.11_env
Executing action: all (aliases: build)
Running ninja in directory C:\Users\aravindsamy\esp\esp-at\build
Executing "ninja all"...
[... cut ...]
C:\Espressif\python_env\idf5.2_py3.11_env\Scripts\python.exe esp-idf\components\esptool_py\esptool\esptool.py -p (PORT) -b 460800 --before default_reset --after hard_reset --chip esp32c3  write_flash --flash_mode dio --flash_size 4MB --flash_freq 40m 0x0 build\bootloader\bootloader.bin 0x8000 build\partition_table\partition-table.bin 0xd000 build\ota_data_initial.bin 0x1e000 build\at_customize.bin 0x1f000 build\customized_partitions\mfg_nvs.bin 0x47000 build\customized_partitions\fatfs.bin 0x60000 build\esp-at.bin
or run 'idf.py -p (PORT) flash' `
```

ustccw, 2024-08-07:
@Aravind0812 thanks for your report! i can reproduce it now. i'll fix it within one week.

closed 2024-08-09 by igrr

ustccw, 2024-08-09:
@Aravind0812 Hi, this issue should be fixed on latest master branch, could you please `git pull` the latest code and have a test, thanks.

Aravind0812, 2024-08-14:
ok i will test. Thank you