espressif/esp-at #844: AT+SYSMSG=0 does not disable WiFi connection related messages

BornaBiro, 2024-06-05:
### Answers checklist.

- [X] I have read the documentation [ESP-AT Programming Guide](https://docs.espressif.com/projects/esp-at/en/latest/) and the issue is not addressed there.
- [X] I have used the latest released firmware or have updated my ESP-AT branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

I'm currently having an issue with the SYSMSG AT command. As described [here](https://docs.espressif.com/projects/esp-at/en/latest/esp32c3/AT_Command_Set/Basic_AT_Commands.html#cmd-sysmsg) setting ATSYSMSG=0 should block all prompt messages from the ESP32 to the master device including the WiFi connection status, but it does not for some reason.

I made a test setup with my phone as a hotspot and ESP32 is connected to my phone. With SYSMSG=0 and ESP32 connected to my phone, I disable the hotspot on my phone and ESP32 immediately sends WIFI DISCONNECTED.

I'm using ESP32-C3 with ESP32 AT Firmware V3.3.0.0 and communication is done with the SPI.

This is a big problem for me since I'm using pass-through mode for getting HTTP GET and HTTP POST.

Here is the AT Command log
```
-----[SENT]-----
AT+SYSMSG=0
---[SENT END]---
-----[RESPONSE]-----
AT+SYSMSG=0

OK

---[RESPONSE END]---
-----[SENT]-----
AT+SYSMSG?
---[SENT END]---
-----[RESPONSE]-----
AT+SYSMSG?
+SYSMSG:0
OK

---[RESPONSE END]---
-----[SENT]-----
AT+CWINIT=1
---[SENT END]---
-----[RESPONSE]-----
AT+CWINIT=1

OK

---[RESPONSE END]---
-----[SENT]-----
AT+CWMODE=1
---[SENT END]---
-----[RESPONSE]-----
AT+CWMODE=1

OK

---[RESPONSE END]---
-----[SENT]-----
AT+CWJAP="testWiFiNetworkEsp32","esp32WiFiPass"
---[SENT END]---
-----[RESPONSE]-----
AT+CWJAP="testWiFiNetworkEsp32","esp32WiFiPass"

---[RESPONSE END]---
-----[RESPONSE]-----
WIFI CONNECTED

---[RESPONSE END]---
-----[RESPONSE]-----
WIFI GOT IP

OK

---[RESPONSE END]---
-----[SENT]-----
AT+SYSMSG?
---[SENT END]---
-----[RESPONSE]-----
AT+SYSMSG?
+SYSMSG:0
OK

---[RESPONSE END]---
-----[RESPONSE]-----
WIFI DISCONNECT       <<<<<<<This!

---[RESPONSE END]---
```

Maybe I'm using a wrong command, so any advice would be helpful.
Thanks!

ustccw, 2024-06-28:
@BornaBiro Hello, you should use [AT+SYSMSGFILTERCFG](https://docs.espressif.com/projects/esp-at/en/latest/esp32/AT_Command_Set/Basic_AT_Commands.html#cmd-sysmsgfiltercfg) to set up system message filters and enable the filter by [AT+SYSMSGFILTER=1](https://docs.espressif.com/projects/esp-at/en/latest/esp32/AT_Command_Set/Basic_AT_Commands.html#cmd-sysmsgfilter). Here is a [detailed example](https://docs.espressif.com/projects/esp-at/en/latest/esp32/AT_Command_Examples/at_msg_resp_fmt_ctrl_examples.html#example-sysmfgfilter).

> `AT+SYSMSG` usually used for [Passthrough Mode](https://docs.espressif.com/projects/esp-at/en/latest/esp32/index_of_abbreviations.html#term-Passthrough-Mode), you can refer to the parameter explanation under the command for more details. In your case, AT works in [Normal Transmission Mode](https://docs.espressif.com/projects/esp-at/en/latest/esp32/index_of_abbreviations.html#term-Normal-Transmission-Mode).

BornaBiro, 2024-07-03:
That was exactly what I used as a workaround in the meantime and it worked perfectly.

One this that I noticed with AT+SYSMSGFILTERCFG is, if there is head and tail regex sent as described [here](https://docs.espressif.com/projects/esp-at/en/latest/esp32c3/AT_Command_Examples/at_msg_resp_fmt_ctrl_examples.html?highlight=passthough#enable-the-system-message-filter-to-download-files-in-http-passthough-mode) for passthrough mode for removing tail and header form the HTTPCGET on every received data, after sending AT+SYSMSGFILTERCFG with it's parameters and waiting for ">", I must send separate SPI transactions for head and tail data, otherwise ESP32 just hangs.

So, if I need to enable filter for the HTTPCGET to remove header and tail from every HTTP data I need to:
1. Send `AT+SYSMSGFILTERCFG=1,18,3\r\n`
2. Wait for response `OK\r\n\r\n>`
3. Send `^+HTTPCGET:[0-9]*,`
4. Wait at least 5 milliseconds
5. Send `\r\n$`

Is this is intended behavior? Because I did not see anywhere documented that head and tail data must be sent separately with some delay in between.

ustccw, 2024-07-03:
@BornaBiro This is not the expected behavior. Could you mind providing your log port logs? 
i would like to reproduce it next week, maybe.

closed 2025-03-17 by ustccw