espressif/esp-at #847: Multiple +BTENCDEV responses not separated with new lines

JHedgesBartec, 2024-06-18:
### Answers checklist

- [X] I have read the documentation [ESP-AT Programming Guide](https://docs.espressif.com/projects/esp-at/en/latest/) and the issue is not addressed there.
- [X] I have used the latest released firmware or have updated my ESP-AT branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### AT+GMR

AT+GMR
AT version:3.2.0.0(s-ec2dec2 - ESP32 - Jul 28 2023 07:05:28)
SDK version:v5.0.2-376-g24b9d38a24-dirty
compile time(5c1d4b67):Jun  7 2024 15:43:17
Bin version:v4.0.0.0(WROOM-32)

OK

### ESP-AT Firmware Source

1. Clone the v3.2.0.0 branch as recommended for the ESP32-WROOM module
2. Modify the SDK config to enable Bluetooth Classic and BT SPP
3. Not required, but change the binary version to show it is different to the branch configuration
4. Build and install using the factory binary

### Hardware Information

ESP32-WROOM32(E)

### Power Supply used

USB

### What is the expected behavior?

When multiple pairings are present, and they are requested using `AT+BTENCDEV?` each pairing response is provided on a new line, for example:
```
AT+BTENCDEV?`
AT+BTENCDEV?
+BTENCDEV:0,"00:07:80:02:47:00"
+BTENCDEV:1,"90:38:0c:ed:67:02"
OK
```

### What is the actual behavior?

```
AT+BTENCDEV?
AT+BTENCDEV?
+BTENCDEV:0,"00:07:80:02:47:00"+BTENCDEV:1,"90:38:0c:ed:67:02"
OK
```

### Probability of recurrence

100% repeatable

### AT+SYSRAM?

```
AT+SYSRAM?
AT+SYSRAM?
+SYSRAM:81624,62892
OK
```

### Steps to reproduce

1. Follow the guide [here](https://docs.espressif.com/projects/esp-at/en/latest/esp32/AT_Command_Examples/classic_bluetooth_at_examples.html#establish-spp-connection-between-two-esp32-development-boards) on how to connect two ESP32 modules. Perform this using identical software.
2. Do this twice to make sure there are two different pairings
3. Send `AT+BTENCDEV?` to request the pairing list

### AT command port output

```plain
AT+BTSPPCONN=0,0,"90:38:0C:ED:67:02"
AT+BTSPPCONN=0,0,"90:38:0C:ED:67:02"
+BTSPPCONN:0,"90:38:0c:ed:67:02"

OK
AT+BTSPPDISCONN=0
AT+BTSPPDISCONN=0
+BTSPPDISCONN:0,"90:38:0c:ed:67:02"

OK
AT+BTSPPCONN=0,0,"00:07:80:02:47:00"
AT+BTSPPCONN=0,0,"00:07:80:02:47:00"
+BTSPPCONN:0,"00:07:80:02:47:00"

OK
AT+BTSPPDISCONN=0
AT+BTSPPDISCONN=0
+BTSPPDISCONN:0,"00:07:80:02:47:00"

OK
AT+BTENCDEV?
AT+BTENCDEV?
+BTENCDEV:0,"00:07:80:02:47:00"+BTENCDEV:1,"90:38:0c:ed:67:02"
OK

```
```

### AT log port output

```plain
No changes to the output log when sending `AT+BTENCDEV?`
```

JHedgesBartec, 2024-06-18:
I have attempted to test this using the latest release of v3.4.0.0 however it does not build successfully with BT Classic SPP enabled for the ESP32-WROOM module (https://github.com/espressif/esp-at/issues/848)

Luxin812, 2024-06-28:
> ### Answers checklist
> * [x]  I have read the documentation [ESP-AT Programming Guide](https://docs.espressif.com/projects/esp-at/en/latest/) and the issue is not addressed there.
> * [x]  I have used the latest released firmware or have updated my ESP-AT branch (master or release) to the latest version and checked that the issue is present there.
> * [x]  I have searched the issue tracker for a similar issue and not found a similar issue.
> 
> ### AT+GMR
> AT+GMR AT version:3.2.0.0(s-ec2dec2 - ESP32 - Jul 28 2023 07:05:28) SDK version:v5.0.2-376-g24b9d38a24-dirty compile time([5c1d4b6](https://github.com/espressif/esp-at/commit/5c1d4b6723b45c619d76c69ced3987ad6f3e3831)):Jun 7 2024 15:43:17 Bin version:v4.0.0.0(WROOM-32)
> 
> OK
> 
> ### ESP-AT Firmware Source
> 1. Clone the v3.2.0.0 branch as recommended for the ESP32-WROOM module
> 2. Modify the SDK config to enable Bluetooth Classic and BT SPP
> 3. Not required, but change the binary version to show it is different to the branch configuration
> 4. Build and install using the factory binary
> 
> ### Hardware Information
> ESP32-WROOM32(E)
> 
> ### Power Supply used
> USB
> 
> ### What is the expected behavior?
> When multiple pairings are present, and they are requested using `AT+BTENCDEV?` each pairing response is provided on a new line, for example:
> 
> ```
> AT+BTENCDEV?`
> AT+BTENCDEV?
> +BTENCDEV:0,"00:07:80:02:47:00"
> +BTENCDEV:1,"90:38:0c:ed:67:02"
> OK
> ```
> 
> ### What is the actual behavior?
> ```
> AT+BTENCDEV?
> AT+BTENCDEV?
> +BTENCDEV:0,"00:07:80:02:47:00"+BTENCDEV:1,"90:38:0c:ed:67:02"
> OK
> ```
> 
> ### Probability of recurrence
> 100% repeatable
> 
> ### AT+SYSRAM?
> ```
> AT+SYSRAM?
> AT+SYSRAM?
> +SYSRAM:81624,62892
> OK
> ```
> 
> ### Steps to reproduce
> 1. Follow the guide [here](https://docs.espressif.com/projects/esp-at/en/latest/esp32/AT_Command_Examples/classic_bluetooth_at_examples.html#establish-spp-connection-between-two-esp32-development-boards) on how to connect two ESP32 modules. Perform this using identical software.
> 2. Do this twice to make sure there are two different pairings
> 3. Send `AT+BTENCDEV?` to request the pairing list
> 
> ### AT command port output
> ```
> AT+BTSPPCONN=0,0,"90:38:0C:ED:67:02"
> AT+BTSPPCONN=0,0,"90:38:0C:ED:67:02"
> +BTSPPCONN:0,"90:38:0c:ed:67:02"
> 
> OK
> AT+BTSPPDISCONN=0
> AT+BTSPPDISCONN=0
> +BTSPPDISCONN:0,"90:38:0c:ed:67:02"
> 
> OK
> AT+BTSPPCONN=0,0,"00:07:80:02:47:00"
> AT+BTSPPCONN=0,0,"00:07:80:02:47:00"
> +BTSPPCONN:0,"00:07:80:02:47:00"
> 
> OK
> AT+BTSPPDISCONN=0
> AT+BTSPPDISCONN=0
> +BTSPPDISCONN:0,"00:07:80:02:47:00"
> 
> OK
> AT+BTENCDEV?
> AT+BTENCDEV?
> +BTENCDEV:0,"00:07:80:02:47:00"+BTENCDEV:1,"90:38:0c:ed:67:02"
> OK
> ```
> 
> ```
> 
> 
> ### AT log port output
> 
> ```plain
> No changes to the output log when sending `AT+BTENCDEV?`
> ```
> 
> ### More Information.
> _No response_

Thank you very much for raising this issue. Yes, AT does indeed have this problem at the moment. We have replicated this issue locally and will soon internally synchronize the fix for this issue to the master branch. Thanks again.

JHedgesBartec, 2024-07-01:
> Thank you very much for raising this issue. Yes, AT does indeed have this problem at the moment. We have replicated this issue locally and will soon internally synchronize the fix for this issue to [...]

Thank you for reviewing this item, do you have a timescale when this will be pushed to a suitable release branch for the ESP32? Master (and v4 and above) is not recommended for the ESP32 module.

Luxin812, 2025-09-09:
You can download the latest released version, ESP32 AT [release v4.1.1.0](https://docs.espressif.com/projects/esp-at/zh_CN/latest/esp32/AT_Binary_Lists/esp_at_binaries.html), for testing. If your issue has been resolved, please close the issue. Thank you!

JHedgesBartec, 2025-09-09:
+BTENCDEV responses are now new line separated

```
AT+GMR
AT version:4.2.0.0-dev(s-39c513e - ESP32 - Sep  1 2025 09:25:18)
SDK version:v5.4.1-643-g8ad0d3d8f2-dirty
compile time(531495c0):Sep  5 2025 14:54:49
Bin version:v5.0.0.0(WROOM-32)

OK
AT+BLEENCDEV?
+BLEENCDEV:0,"4c:fa:e6:fa:f7:a4"
+BLEENCDEV:1,"d4:54:8b:fd:6d:0c"

OK
```

closed 2025-09-09 by JHedgesBartec