espressif/esp-at #848: ESP-AT does not build for ESP32-WROOM with BT Classic enabled on release v3.4.0.0

JHedgesBartec, 2024-06-18:
### Answers checklist

- [X] I have read the documentation [ESP-AT Programming Guide](https://docs.espressif.com/projects/esp-at/en/latest/esp32/Compile_and_Develop/How_to_clone_project_and_compile_it.html) and the issue is not addressed there.
- [X] I have compiled the hello_world example [Get Started with ESP-IDF](https://docs.espressif.com/projects/esp-at/en/latest/esp32/Compile_and_Develop/How_to_clone_project_and_compile_it.html#step-1-get-started-with-esp-idf) and the issue is not addressed there.
- [X] I have updated my ESP-AT branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### ESP-AT version

v3.4.0.0

### Operating System used

Linux

### If you are using Windows, please specify command line type.

None

### What is the expected behavior?

Application should build successfully when BT Classic and SPP support is enabled as per the 'Important' note [here](https://docs.espressif.com/projects/esp-at/en/release-v3.4.0.0/esp32/AT_Command_Set/BT_AT_Commands.html)

- [ ] 

### What is the actual behavior?

When Bluetooth Classic and SPP is enabled, the application does not build with an error saying application is too large for the ota partitions.
In our application we do not need OTA, however disabling OTA through menuconfig does not change what partition table is used or the factory binary to not use the OTA partitions.
Having a way to fully remove OTA support may be an acceptable resolution to this.

### Steps to reproduce

1. Clone branch v3.4.0.0
2. Configure for ESP32-WROOM-32 with Silence Mode enabled
3. As per the important note [here ](https://docs.espressif.com/projects/esp-at/en/release-v3.4.0.0/esp32/AT_Command_Set/BT_AT_Commands.html)enable BT Classic SPP support:
3a. Enable `Component Config -> Bluetooth -> Bluedroid Options -> Enable SPP`
3b. Enable `Component Config -> AT-> AT bt command support` and ensure that SPP is also enabled
4. Build using the build script
5. Build fails with `Error: All app partitions are too small for binary esp-at.bin size`

### Build or installation Logs

```plain
Platform name:ESP32	Module name:WROOM-32	Silence:1
submodules check completed for updates.
esp_tls_parse_ecc_key.patch does not need to be applied.
fatfs_generation.patch does not need to be applied.
support_ext_partition.patch does not need to be applied.
modify_bluedroid_conn_timeout.patch does not need to be applied.
/home/bartec/Development/working_directory/esp-at/module_config/module_esp32_default/patch/libbtdm_app.a does not need to be applied.
patches check completed for updates.
Ready to set up environment variables..
PATH is /home/bartec/.local/bin:/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin:/usr/games:/usr/local/games:/snap/bin:/snap/bin
IDF_PYTHON_ENV_PATH is None
sys.platform is linux
export str is OPENOCD_SCRIPTS=/home/bartec/.espressif/tools/openocd-esp32/v0.12.0-esp32-20230921/openocd-esp32/share/openocd/scripts
ESP_ROM_ELF_DIR=/home/bartec/.espressif/tools/esp-rom-elfs/20220823/
IDF_PYTHON_ENV_PATH=/home/bartec/.espressif/python_env/idf5.0_py3.10_env
ESP_IDF_VERSION=5.0
PATH=/home/bartec/.espressif/tools/xtensa-esp-elf-gdb/12.1_20231023/xtensa-esp-elf-gdb/bin:/home/bartec/.espressif/tools/xtensa-esp32-elf/esp-2022r1-11.2.0/xtensa-esp32-elf/bin:/home/bartec/.espressif/tools/esp32ulp-elf/2.35_20220830/esp32ulp-elf/bin:/home/bartec/.espressif/tools/openocd-esp32/v0.12.0-esp32-20230921/openocd-esp32/bin:/home/bartec/.espressif/python_env/idf5.0_py3.10_env/bin:/home/bartec/Development/working_directory/esp-at/esp-idf/tools:$PATH
IDF_DEACTIVATE_FILE_PATH=/tmp/tmp_bhx7frbidf_97403

PATH is /home/bartec/.espressif/tools/xtensa-esp-elf-gdb/12.1_20231023/xtensa-esp-elf-gdb/bin:/home/bartec/.espressif/tools/xtensa-esp32-elf/esp-2022r1-1
1.2.0/xtensa-esp32-elf/bin:/home/bartec/.espressif/tools/esp32ulp-elf/2.35_20220830/esp32ulp-elf/bin:/home/bartec/.espressif/tools/openocd-esp32/v0.12.0-
esp32-20230921/openocd-esp32/bin:/home/bartec/.espressif/python_env/idf5.0_py3.10_env/bin:/home/bartec/Development/working_directory/esp-at/esp-idf/tools
:$PATH:/home/bartec/.local/bin:/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin:/usr/games:/usr/local/games:/snap/bin:/snap/bin
IDF_PYTHON_ENV_PATH is /home/bartec/.espressif/python_env/idf5.0_py3.10_env
Executing action: all (aliases: build)
Running ninja in directory /home/bartec/Development/working_directory/esp-at/build
Executing "ninja all"...
[1/1] cd /home/bartec/Development/working_directory/esp-at/build/bootloader...bartec/Development/working_directory/esp-at/build/bootloader/bootloader.bin
Bootloader binary size 0x6330 bytes. 0xcd0 bytes (11%) free.
[3/5] Running utility command for customized_bin
Creating NVS binary with version: V2 - Multipage Blob Support Enabled

Created NVS binary: ===> /home/bartec/Development/working_directory/esp-at/build/customized_partitions/mfg_nvs.bin
[4/5] cd /home/bartec/Development/working_directory/esp-at/build/esp-idf/es...able.bin /home/bartec/Development/working_directory/esp-at/build/esp-at.bin
FAILED: esp-idf/esptool_py/CMakeFiles/app_check_size /home/bartec/Development/working_directory/esp-at/build/esp-idf/esptool_py/CMakeFiles/app_check_size
 
cd /home/bartec/Development/working_directory/esp-at/build/esp-idf/esptool_py && /home/bartec/.espressif/python_env/idf5.0_py3.10_env/bin/python /home/ba
rtec/Development/working_directory/esp-at/esp-idf/components/partition_table/check_sizes.py --offset 0x8000 partition --type app /home/bartec/Development
/working_directory/esp-at/build/partition_table/partition-table.bin /home/bartec/Development/working_directory/esp-at/build/esp-at.bin
Error: All app partitions are too small for binary esp-at.bin size 0x1825d0:
  - Part 'ota_0' 0/16 @ 0x100000 size 0x180000 (overflow 0x25d0)
  - Part 'ota_1' 0/17 @ 0x280000 size 0x180000 (overflow 0x25d0)
ninja: build stopped: subcommand failed.
ninja failed with exit code 1, output of the command is in the /home/bartec/Development/working_directory/esp-at/build/log/idf_py_stderr_output_97445 and
 /home/bartec/Development/working_directory/esp-at/build/log/idf_py_stdout_output_97445
A fatal error occurred: idf.py build failed
```

### More Information

[idf_py_stderr_output_97445.log](https://github.com/user-attachments/files/15887334/idf_py_stderr_output_97445.log)
[idf_py_stdout_output_97445.log](https://github.com/user-attachments/files/15887335/idf_py_stdout_output_97445.log)

Luxin812, 2024-06-28:
`Error: All app partitions are too small for binary esp-at.bin size 0x1825d0:` 
The compilation failure is primarily due to enabling Bluetooth functionality, resulting in the esp-at.bin file being too large. You can reduce the size of esp-at.bin during compilation by disabling some AT functionalities that you do not use.

JHedgesBartec, 2024-07-31:
I have managed to get this to build with the following changes:

- Ethernet ->
   - Disable support SPI to ethernet module
- ESP HTTPS OTA ->
   - Disable allow HTTP for OTA
- AT ->
   - Disable WPS support
   - Enable AT bt command support with SPP (default)
   - Disable OTA support
   - Disable User command support
   - Disable SmartConfig command support
   - Disable Signalling command support

closed 2024-07-31 by JHedgesBartec