espressif/esp-at #853: When using Ethernet, esp32 wroom cannot enter sleep mode.

charon22, 2024-07-10:
### Answers checklist.

- [X] I have read the documentation [ESP-AT Programming Guide](https://docs.espressif.com/projects/esp-at/en/latest/) and the issue is not addressed there.
- [X] I have used the latest released firmware or have updated my ESP-AT branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

ESP-AT version
v3.4.0.0

If I enable Ethernet and send a sleep command, it does not enter sleep.
If I build without Ethernet and send the sleep command, it will enter sleep mode normally.
sdkconfig

CONFIG_AT_ETHERNET_SUPPORT=y
CONFIG_PHY_RTL8201=y

I would like to know how to enter sleep mode when using Ethnet.

Thanks
Sungsu

ustccw, 2024-10-10:
@charon22 We sincerely apologize that [esp-idf](https://github.com/espressif/esp-idf) has not yet optimized the power consumption for Ethernet usage. This means that when Ethernet is enabled, the module may not be able to enter light-sleep, resulting in relatively higher power consumption.

If this feature is essential for your product, we kindly suggest submitting detailed requirements to [esp-idf issue](https://github.com/espressif/esp-idf/issues/new?assignees=&labels=Type%3A+Feature+Request&projects=&template=03_feature_request.yml).

ustccw, 2024-10-10:
same to https://github.com/espressif/esp-at/issues/832

closed 2024-10-10 by charon22