espressif/esp-at #855: esp32 c3 wroom 02  run to at commands with doclight  (IDFGH-13240)

russianbearx, 2024-07-15:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

ESP32-C3-MINI-1-AT-V2.2.0.0

### Operating System used.

Windows

### How did you build your project?

Command line with Make

### If you are using Windows, please specify command line type.

CMD

### What is the expected behavior?

I want to run the esp32c3 wroom 02 card with at commands with the docklight application, but I get the error "invalid header 0xffffff<CR><LF>". can you help?

### What is the actual behavior?

I want to run the esp32c3 wroom 02 card with at commands with the docklight application. in the simplest form, I want to get the OK command when I run the AT command.

### Steps to reproduce.

PS C:\Users\oguzh\Desktop\at firmware icin\ESP32-C3-MINI-1-V3.2.0.0> esptool.exe --chip esp32c3 --port COM3 --baud 115200 write_flash -z 0x1000 "C:\Users\oguzh\Desktop\at firmware icin\ESP32-C3-MINI-1-V3.2.0.0\bootloader\bootloader.bin" 0x8000 "C:\Users\oguzh\Desktop\at firmware icin\ESP32-C3-MINI-1-V3.2.0.0\partition_table\partition-table.bin" 0x10000 "C:\Users\oguzh\Desktop\at firmware icin\ESP32-C3-MINI-1-V3.2.0.0\at firmware icin.bin" 0xd000 "C:\Users\oguzh\Desktop\at firmware icin\ESP32-C3-MINI-1-V3.2.0.0\ota_data_initial.bin"
esptool.py v4.7.0
Serial port COM3
Connecting....
Chip is ESP32-C3 (QFN32) (revision v0.4)
Features: WiFi, BLE
Crystal is 40MHz
MAC: 58:cf:79:27:64:e4
Uploading stub...
Running stub...
Stub running...
Configuring flash size...
Flash will be erased from 0x00001000 to 0x00005fff...
Flash will be erased from 0x00008000 to 0x00008fff...
Flash will be erased from 0x00010000 to 0x001b0fff...
Flash will be erased from 0x0000d000 to 0x0000efff...
Compressed 20240 bytes to 12435...
Wrote 20240 bytes (12435 compressed) at 0x00001000 in 1.5 seconds (effective 105.8 kbit/s)...
Hash of data verified.
Compressed 3072 bytes to 131...
Wrote 3072 bytes (131 compressed) at 0x00008000 in 0.1 seconds (effective 249.5 kbit/s)...
Hash of data verified.
Compressed 1707760 bytes to 969290...
Wrote 1707760 bytes (969290 compressed) at 0x00010000 in 86.0 seconds (effective 158.9 kbit/s)...
Hash of data verified.
Compressed 8192 bytes to 31...
Wrote 8192 bytes (31 compressed) at 0x0000d000 in 0.1 seconds (effective 497.7 kbit/s)...
Hash of data verified.

Leaving...
Hard resetting via RTS pin...
PS C:\Users\oguzh\Desktop\at firmware icin\ESP32-C3-MINI-1-V3.2.0.0>

russianbearx, 2024-07-15:
I have examined all of the links you suggested me to look at in detail and I have done all the desired things here. when I connect my esp32 device with ttl, the led connected to the RX pin lights up without stopping. so when I send the "AT" command in the docklight program, I keep getting the "invalid header 0xffffff" error. i would be very grateful if you could help me a little more as I am new to these jobs. what I want to do is to get the "OK" response exactly in the screenshot you sent.

Luxin812, 2024-07-15:
@russianbearx The log below shows that you have only burned four firmwares.
```PS C:\Users\oguzh\Desktop\at firmware icin\ESP32-C3-MINI-1-V3.2.0.0> esptool.exe --chip esp32c3 --port COM3 --baud 115200 write_flash -z 0x1000 "C:\Users\oguzh\Desktop\at firmware icin\ESP32-C3-MINI-1-V3.2.0.0\bootloader\bootloader.bin" 0x8000 "C:\Users\oguzh\Desktop\at firmware icin\ESP32-C3-MINI-1-V3.2.0.0\partition_table\partition-table.bin" 0x10000 "C:\Users\oguzh\Desktop\at firmware icin\ESP32-C3-MINI-1-V3.2.0.0\at firmware icin.bin" 0xd000 "C:\Users\oguzh\Desktop\at firmware icin\ESP32-C3-MINI-1-V3.2.0.0\ota_data_initial.bin"```
 Please confirm whether you have correctly Download the AT firmware into the flash according to [this document](https://docs.espressif.com/projects/esp-at/en/latest/esp32c3/Get_Started/Downloading_guide.html).
```--flash_mode dio --flash_freq 40m --flash_size 4MB
0x0 bootloader/bootloader.bin
0x8000 partition_table/partition-table.bin
0xd000 ota_data_initial.bin
0x1e000 at_customize.bin
0x1f000 customized_partitions/mfg_nvs.bin
0x60000 esp-at.bin```

closed 2024-10-24 by ustccw