espressif/esp-at #856: +CWLAP Response contains brackets which break the API convention

JHedgesBartec, 2024-07-19:
### Answers checklist

- [X] I have read the documentation [ESP-AT Programming Guide](https://docs.espressif.com/projects/esp-at/en/latest/) and the issue is not addressed there.
- [X] I have used the latest released firmware or have updated my ESP-AT branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### AT+GMR

Source built from v3.2.0.0 release branch with BT Classic enabled, hence the modified BIN version.

```
AT+GMR
AT+GMR
AT version:3.2.0.0(s-ec2dec2 - ESP32 - Jul 28 2023 07:05:28)
SDK version:v5.0.2-376-g24b9d38a24-dirty
compile time(5c1d4b67):Jul 16 2024 17:06:45
Bin version:v4.0.0.0(WROOM-32)

OK
```

### ESP-AT Firmware Source

v3.2.0.0 release branch, compiled locally to enable BT Classic on the ESP32-WROOM-32

### Hardware Information

ESP32-WROOM-32

### Power Supply used

USB

### What is the expected behavior?

The +CWLAP response conforms to the documented format [here](https://docs.espressif.com/projects/esp-at/en/release-v3.2.0.0/esp32/AT_Command_Set/Wi-Fi_AT_Commands.html#cmd-lap)

`+CWLAP:<ecn>,<ssid>,<rssi>,<mac>,<channel>,<freq_offset>,<freqcal_val>,<pairwise_cipher>,<group_cipher>,<bgn>,<wps>
`

### What is the actual behavior?

The +CWLAP response contains () brackets around all parameters. This does not match the documentation and breaks the general API format of <cmd>: followed by comma separated parameters. Having the parameters bracketed means it does not conform to the format of other API commands and has to be custom parsed.

Example:
`+CWLAP:(3,"TEST_NETGEAR26",-54,"22:6d:34:30:7f:72",11,-1,-1,4,4,7,0)
`

### Probability of recurrence

100% repeatable

### AT+SYSRAM?

AT+SYSRAM?
AT+SYSRAM?
+SYSRAM:151140,144872
OK

### Steps to reproduce

Initialise Wi-Fi in station mode with `AT+CWMODE=1` and then perform a list of access points using `AT+CWLAP`

### AT command port output

```plain
AT+GMR
AT+GMR
AT version:3.2.0.0(s-ec2dec2 - ESP32 - Jul 28 2023 07:05:28)
SDK version:v5.0.2-376-g24b9d38a24-dirty
compile time(5c1d4b67):Jul 16 2024 17:06:45
Bin version:v4.0.0.0(WROOM-32)

OK
AT+CWMODE=1
AT+CWMODE=1

OK
AT+CWLAP
AT+CWLAP
+CWLAP:(3,"TEST_NETGEAR26",-52,"22:6d:34:30:7f:72",11,-1,-1,4,4,7,0)

OK
AT+SYSRAM?
AT+SYSRAM?
+SYSRAM:151140,144872
OK
```

### AT log port output

```plain
Compiled with silence mode so no log output.
```

JAndrassy, 2024-07-19:
it is this way a very long time now. it would be a breaking change if the {} are removed

JHedgesBartec, 2024-07-31:
Authorised personnel should then update the documentation to reflect actual behaviour

ustccw, 2024-08-07:
@JHedgesBartec @JAndrassy Thanks a lot! We should update the ESP-AT document.

ustccw, 2025-09-09:
The documentation was updated a long time ago, thank you again!

closed 2025-09-09 by ustccw