espressif/esp-at #857: BLE Scan Window Ignored, BLESCANPARAM

Peter-B4321, 2024-07-22:
### Answers checklist

- [X] I have read the documentation [ESP-AT Programming Guide](https://docs.espressif.com/projects/esp-at/en/latest/esp32/Compile_and_Develop/How_to_clone_project_and_compile_it.html) and the issue is not addressed there.
- [X] I have compiled the hello_world example [Get Started with ESP-IDF](https://docs.espressif.com/projects/esp-at/en/latest/esp32/Compile_and_Develop/How_to_clone_project_and_compile_it.html#step-1-get-started-with-esp-idf) and the issue is not addressed there.
- [X] I have updated my ESP-AT branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### ESP-AT version

3.4.0.0

### Operating System used

Linux

### If you are using Windows, please specify command line type.

None

### What is the expected behavior?

Scan for the window and wait the interval.
For example, AT+BLESCANPARAM=1,0,0,16000,1600
The command should scan for 1s and then wait 9s.

### What is the actual behavior?

Scanning is constant without breaks.

### Steps to reproduce

AT+BLEINIT=1
AT+BLESCANPARAM=1,0,0,16000,1760
AT+BLESCAN=1,0
Then, look at the scan results.

### Build or installation Logs

```plain
This issue is new to ESP-AT v3.4.0.0
In v3.2.0.0, the ESP would scan for 1 second and wait 9, then repeat.
The behavior seems like the window is equal to the interval (AT+BLESCANPARAM=1,0,0,1760,1760)
```

### More Information

Happens on multiple devices.

Peter-B4321, 2024-07-22:
AT+GMR response:
AT version:3.4.0.0(s-c31b833 - ESP32 - Jun  7 2024 03:48:17)
SDK version:v5.0.6-dirty
compile time(70ff5889):Jun  7 2024 04:46:00
Bin version:v3.4.0.0(WROOM-32)
OK

JHedgesBartec, 2024-08-01:
I have also seen this issue on v3.2.0.0 with the ESP-WROOM-32

It is clear that both the scan interval and scan window are only applied when the BLE module is de-initialised (`AT+BLEINIT=0`) then initialised as a client again (`AT+BLEINIT=1`) or the module is rebooted (`AT+RST`) and then BLE re-initialised.

Peter-B4321, 2024-08-05:
On v3.4.0.0, applying the scan parameters while BLE is de-initialized returns an ERROR

TX: AT+BLEINIT=0
RX: OK
TX: AT+BLESCANPARAM=1,0,0,16000,1760
RX: ERROR
TX: AT+BLEINIT=1
RX: OK
TX: AT+BLESCANPARAM=1,0,0,16000,1760
RX: OK

JHedgesBartec, 2024-08-06:
@Peter-B4321 to work around this issue you will need to apply the new parameters, then de-initialise BLE before initialising it again.