espressif/esp-at #858: Add support for BLE SPP normal transmission mode

JHedgesBartec, 2024-07-25:
### Is your feature request related to a problem?

Both BT Classic and Wi-FI have support for normal mode when using the respective send parameters. When maintaining multiple connections having normal mode is beneficial.
Switching into passthrough mode for BLE SPP makes handling BT/Wi-Fi data difficult.

### Describe the solution you'd like.

A method of sending data to a BLE SPP connection in normal mode, where once the data is transmitted to the ESP32 module, the module returns OK and returns to normal command mode.
For example adding length to the `AT+BLESPP` command or a new send command such as `AT+BLESPPSEND=<length>` which could work similar to `AT+BTSPPSEND` and `AT+CIPSEND`

### Describe alternatives you've considered.

No other approach.

JavierReyes945, 2024-07-30:
@JHedgesBartec:

I recently worked this situation myself, and discovered that you can use the SPP without activating the passthrough mode. 

What I do is to configure SPP as normal (based on the example code)

```
- AT+BLESPPCFG=1,1,7,1,5
```

But i **do not activate** the  SPP Mode (I do not enter `AT+BLESPP` command).

When the connection is made, I will receive data from the remote device written to the RX channel (characteristic 304 registered for RX in SPP in the examples) as:

```
+WRITE:0,1,5,,<len>,<data>\r\n
```

And when we want to send, we use the characteristic registered for TX (it has to be property notify or indicate according to documentation, so in our case we use the 306, which has been subscribed in remote device):

```
AT+BLEGATTSIND=0,1,7,<len>\r\n
```

After this command, we receive 1 byte (the prompt symbol `>`), then we send the raw data (as many bytes as in <len> announced, and we receive a confirmation message `\r\OK\r\n`.

It works for us on v4.0.0.0/release and in master branch of the ESP-AT firmware (or device is ESP32-C6-MINI-1). It is not documented to work like it, so it would be better if we get a clear answer from Espressif for this, but for now it works for us and might work for you.

JHedgesBartec, 2024-07-31:
@JavierReyes945
Thank you for taking the time to respond and sharing your solution, we will certainly look at using that approach.

As you say it would be good to get an answer from Espressif for this, I will leave the issue open until then.

ruoxiaoye, 2025-02-08:
A more elegant way to solve this problem is to add CMUX support!

Luxin812, 2025-03-26:
@JHedgesBartec 
Currently, the normal transmission mode in BLE AT uses the AT+BLEGATTCWR, AT+BLEGATTSNTFY, and AT+BLEGATTSIND commands. You can refer to the following [example](https://docs.espressif.com/projects/esp-at/en/latest/esp32/AT_Command_Examples/bluetooth_le_at_examples.html#bluetooth-le-client-reads-and-write-services). At present, the AT+BLESPP command does not support normal transmission mode. However, we greatly appreciate your suggestion, and we will discuss it to evaluate its feasibility at the protocol level, as well as consider other factors to determine whether to support this request.