espressif/esp-at #859: Persistent Bonding not working consistently on ESP32-C6-DevKitM-1

JavierReyes945, 2024-07-26:
### Answers checklist

- [X] I have read the documentation [ESP-AT Programming Guide](https://docs.espressif.com/projects/esp-at/en/latest/) and the issue is not addressed there.
- [X] I have used the latest released firmware or have updated my ESP-AT branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### AT+GMR

Due to the fact that the NimBLE config for persistent storage of the bonding information is not active in the release v4.0.0.0, we had to generate a custom build based of master branch, where this setting is active. It is strange that the persistent storage of bonding data is not active by default in the release version...

```
ESP-ROM:esp32c6-20220919
Build:Sep 19 2022
rst:0x1 (POWERON),boot:0x6f (SPI_FAST_FLASH_BOOT)
SPIWP:0xee
mode:DIO, clock div:1
load:0x4086c410,len:0xd2c
load:0x4086e610,len:0x2df0
load:0x40875728,len:0x17d8
SHA-256 comparison failed:
Calculated: f356853ee9de5b48f66989025102f1b96e1605b5ce06fd144e40f80d57e2ad70
Expected: 6e517dcbb71f9e1f11e7d0c897cbb9db17ec7e484f5cf7af3860027794b50b1d
Attempting to boot anyway...
entry 0x4086c410
I (42) boot: ESP-IDF v5.1.2-dirty 2nd stage bootloader
I (42) boot: compile time Jul 23 2024 17:35:26
I (43) boot: chip revision: v0.0
I (45) boot.esp32c6: SPI Speed      : 80MHz
I (50) boot.esp32c6: SPI Mode       : DIO
I (55) boot.esp32c6: SPI Flash Size : 4MB
I (60) boot: Enabling RNG early entropy source...
I (65) boot: Partition Table:
I (69) boot: ## Label            Usage          Type ST Offset   Length
I (76) boot:  0 otadata          OTA data         01 00 0000d000 00002000
I (83) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (91) boot:  2 nvs              WiFi data        01 02 00010000 0000e000
I (98) boot:  3 at_customize     unknown          40 00 0001e000 00042000
I (106) boot:  4 ota_0            OTA app          00 10 00060000 001d0000
I (113) boot:  5 ota_1            OTA app          00 11 00230000 001d0000
I (121) boot: End of partition table
I (125) esp_image: segment 0: paddr=00060020 vaddr=42150020 size=39080h (233600) map
I (182) esp_image: segment 1: paddr=000990a8 vaddr=40800000 size=06f70h ( 28528) load
I (189) esp_image: segment 2: paddr=000a0020 vaddr=42000020 size=14ef98h (1372056) map
I (473) esp_image: segment 3: paddr=001eefc0 vaddr=40806f70 size=12af4h ( 76532) load
I (491) esp_image: segment 4: paddr=00201abc vaddr=40819a70 size=03bd4h ( 15316) load
I (495) esp_image: segment 5: paddr=00205698 vaddr=50000000 size=00068h (   104) load
I (503) boot: Loaded app from partition at offset 0x60000
I (503) boot: Disabling RNG early entropy source...
no external 32k oscillator, disable it now.
at param mode: 1
AT cmd port:uart1 tx:7 rx:6 cts:5 rts:4 baudrate:115200
module_name: ESP32C6-4MB
max tx power=78, ret=0
v4.1.0.0-dev
controller lib commit: [77d09ce]
```

### ESP-AT Firmware Source

Official release bin (factory), as well as generated in Github Actions and also locally compiled binary

### Hardware Information

ESP32-C6-DevKitM-1

### Power Supply used

USB

### What is the expected behavior?
[... 466 lines cut ...]
[2024-07-26 10:58:05.006] registering descriptor 0x2901 with handle=16
[2024-07-26 10:58:05.012] registering characteristic 0xc302 with def_handle=17 val_handle=18
[2024-07-26 10:58:05.017] registering descriptor 0x2901 with handle=19
[2024-07-26 10:58:05.023] registering characteristic 0xc303 with def_handle=20 val_handle=21
[2024-07-26 10:58:05.028] registering descriptor 0x2901 with handle=22
[2024-07-26 10:58:05.034] registering characteristic 0xc304 with def_handle=23 val_handle=24
[2024-07-26 10:58:05.040] registering characteristic 0xc305 with def_handle=25 val_handle=26
[2024-07-26 10:58:05.045] registering characteristic 0xc306 with def_handle=28 val_handle=29
[2024-07-26 10:58:05.051] registering characteristic 0xc307 with def_handle=31 val_handle=32
[2024-07-26 10:58:05.056] registering descriptor 0x2901 with handle=33
[2024-07-26 10:58:05.062] registered service 0xa003 with handle=34
[2024-07-26 10:58:05.062] registering characteristic 0xc400 with def_handle=35 val_handle=36
[2024-07-26 10:58:05.067] registering descriptor 0x2901 with handle=37
[2024-07-26 10:58:05.073] registering characteristic 0xc401 with def_handle=38 val_handle=39
[2024-07-26 10:58:05.079] registering descriptor 0x2901 with handle=40
[2024-07-26 10:58:05.083] reset reason=0
[2024-07-26 10:59:21.856] connection established; status=0 handle=1 our_ota_addr_type=0 our_ota_addr=40849eaf:00:15:40828000:00:42185d34
[2024-07-26 10:59:21.862]  our_id_addr_type=0 our_id_addr=40849ea1:00:15:40828000:00:42185d34
[2024-07-26 10:59:21.867]  peer_ota_addr_type=1 peer_ota_addr=40849eb6:00:15:40828000:00:42185d34
[2024-07-26 10:59:21.873]  peer_id_addr_type=1 peer_id_addr=40849ea8:00:15:40828000:00:42185d34
[2024-07-26 10:59:21.884]  conn_itvl=24 conn_latency=0 supervision_timeout=500 encrypted=0 authenticated=0 bonded=0
[2024-07-26 10:59:22.117] advertise complete; reason=0mtu update event; conn_handle=1 cid=4 mtu=203
[2024-07-26 10:59:32.286] subscribe event; conn_handle=1 attr_handle=29 reason=1 prevn=0 curn=0 previ=0 curi=1
```
```

JavierReyes945, 2024-07-26:
Attaching LOG file from ESP as the issue template did not allow it as text:
[test_esp_pairing.txt](https://github.com/user-attachments/files/16389806/test_esp_pairing.txt)

XionHsieh, 2024-09-04:
We will fix the issue as soon as possible and get back to you. Thank you for your patience.

Luxin812, 2024-10-18:
@JavierReyes945 The issue of encrypted pairing information not being saved in NVS has been resolved. You can download the latest master branch AT firmware for testing to confirm if the problem is fixed.

JavierReyes945, 2024-10-31:
> @JavierReyes945 The issue of encrypted pairing information not being saved in NVS has been resolved. You can download the latest master branch AT firmware for testing to confirm if the problem is fixed.

Thanks for the feedback, initial testing shows good results, we are currently testing in our end product. As soon as those tests are finished, we will update/close the ticket.