espressif/esp-at #860: AT+HTTPCGET binary mode problem with newlines

ringtailsoftware, 2024-08-02:
### Answers checklist.

- [X] I have read the documentation [ESP-AT Programming Guide](https://docs.espressif.com/projects/esp-at/en/latest/) and the issue is not addressed there.
- [X] I have used the latest released firmware or have updated my ESP-AT branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

Using the method from https://github.com/espressif/esp-at/blob/84b025894debf0d3fb025d245c7377f45328de44/docs/en/AT_Command_Examples/at_msg_resp_fmt_ctrl_examples.rst I am attempting to download a binary file.

It appears to work for some files and not others and it seems that newline characters in the middle of a file cause the transfer to end early. Is there a way to disable whatever it's doing with newlines and get a true binary passthrough of data?

I have a webserver with a file containing the following 9 bytes 0d0a0d0a0d0a0a0d0a ("\r\n\r\n\r\n\n\r\n"). On downloading I get:

```
0d0a
0d0a
0d0a4f4b0d0a ("\r\nOK\r\n")

```
(For more information see my forum post at https://www.esp32.com/viewtopic.php?f=42&t=38650&p=135583#p135583)

ustccw, 2024-08-07:
@ringtailsoftware Thanks for your report! i'll fix it within one week.

closed 2024-08-16 by igrr

ustccw, 2024-08-16:
@ringtailsoftware Hello, this issue should get fixed on master branch, could you please `git pull` the code and have a test.

ringtailsoftware, 2024-08-16:
Thank you. This is working in all of my tests so far.

ustccw, 2024-08-16:
Glad to hear that! If you have any other questions or need further assistance, please feel free to let us know.