espressif/esp-at #861: No response when using Battery power supply

Xellosi, 2024-08-04:
### Answers checklist.

- [X] I have read the documentation [ESP-AT Programming Guide](https://docs.espressif.com/projects/esp-at/en/latest/) and the issue is not addressed there.
- [X] I have used the latest released firmware or have updated my ESP-AT branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

Hardware: ESP32-WROOM-32D
ESP-AT Firmware Source: Release v3.4.0.0

I am trying to integrate the ESP32 AT suite with an STM32 development board via UART. When the ESP32 board is powered by the PC's USB port, it responds correctly to commands sent from both the STM32 board and a serial communication tool on the PC. However, when I change the USB connection to a 18650 battery-powered power bank, which output specs is 5 volts and 1 amp, there is no response after sending commands.

Another approach was attempted by changing the power source to a breadboard power supply module, using wires to connect the 5V, 3.3V, and GND pins on the ESP32-WROOM board, but the board still did not send any responses.

I would like to ask if there is a connection error or how to troubleshoot this issue.

ustccw, 2024-08-06:
@Xellosi Hi, it looks like a hardware-related issue. Could you measure the voltage between the VCC and GND pins of the ESP32 device to see if it is stable at 3.3V?

Xellosi, 2024-08-07:
> @Xellosi Hi, it looks like a hardware-related issue. Could you measure the voltage between the VCC and GND pins of the ESP32 device to see if it is stable at 3.3V?

I don't have a multimeter currently. I will try to get one and do measurements, then reply. Thanks.

Xellosi, 2025-01-05:
Sorry for the late response. I wanted to resolve this issue on my own. Following my previous update, I decided to connect the other modules to the power supply of the STM32 board. I also tried connecting the aforementioned power bank to the USB port on the ESP32 board and measured the pin voltage, which was around 5.03V.

closed 2025-01-05 by Xellosi