espressif/esp-at #877: esp32c6 at,loop ets_loader.c 68

staunchheart, 2024-09-30:
### Is your feature request related to a problem?

esp32c6 at 
[image]
 run log
ESP-ROM:esp32c6-20220919
Build:Sep 19 2022
rst:0x1 (POWERON)üÑé0xc (SPI_FASüESP-ROM:esp32c6-20220919
Build:Sep 19 2022
rst:0x1 (POWERON),boot:0xc (SPI_FAST_FLASH_BOOT)
SPIWP:0xee
mode:DIO, clock div:1
load:0x4086c410,len:0xd2c
load:0x4086e610,len:0x2df0
load:0x40875728,len:0x17d8
SHA-256 comparison failed:
Calculated: d65693cff09e7b367f989d125acd3b3f96d703194a64c8151b768efc9e4b6ad4
Expected: c7090fbe9465c3e32c3100bdb5f14c08f0027429a13db017338daf29c9202110
Attempting to boot anyway...
entry 0x4086c410
[0;32mI (42) boot: ESP-IDF v5.1ESP-ROM:esp32c6-20220919
BESP-ROM:esp32c6-20220919
Build:Sep 19 2022
rst:0x1 (POWERON),boot:0xc (SPI_FAST_FLASH_BOOT)
SPIWP:0xee
mode:DIO, clock div:1
load:0x4086c410,len:0xd2c
load:0x4086e610,len:0x2df0
load:0x40875728,len:0x17d8
SHA-256 comparison failed:
Calculated: d65693cff09e7b367f989d125acd3b3f96d703194a64c8151b768efc9e4b6ad4
Expected: c7090fbe9465c3e32c3100bdb5f14c08f0027429a13db017338daf29c9202110
Attempting to boot anyway...
entry 0x4086c410
[0;32mI (42) boot: ESP-IDF v5.1.2-dirty 2nd stage bootloader[0m
[0;32mI (42) boot: compile time Jan 2 2024 05:53:08[0m
[0;32mI (43) boot: chip revision: v0.1[0m
[0;32mI (45) boot.esp32c6: SPI Speed : 80MHz[0m
[0;32mI (50) boot.esp32c6: SPI Mode : DIO[0m
[0;32mI (55) boot.esp32c6: SPI Flash Size : 4MB[0m
[0;32mI (60) boot: Enabling RNG early entropy source...[0m
[0;32mI (65) boot: Partition Table:[0m
[0;32mI (69) boot: ## Label Usage Type ST Offset Length[0m
[0;32mI (76) boot: 0 otadata OTA data 01 00 0000d000 00002000[0m
[0;32mI (83) boot: 1 phy_init RF data 01 01 0000f000 00001000[0m
[0;32mI (91) boot: 2 nvs WiFi data 01 02 00010000 0000e000[0m
[0;32mI (98) boot: 3 at_customize unknown 40 00 0001e000 00042000[0m
[0;32mI (106) boot: 4 ota_0 OTA app 00 10 00060000 001d0000[0m
[0;32mI (113) boot: 5 ota_1 OTA app 00 11 00230000 001d0000[0m
[0;32mI (121) boot: End of partition table[0m
[0;32mI (125) esp_image: segment 0: paddr=00060020 vaddr=42150020 size=30740h (198464) map[0m
[0;32mI (175) esp_image: segment 1: paddr=00090768 vaddr=40800000 size=0f8b0h ( 63664) load[0m
[0;32mI (190) esp_image: segment 2: paddr=000a0020 vaddr=42000020 size=146000h (1335296) map[0m
[0;32mI (466) esp_image: segment 3: paddr=001e6028 vaddr=4080f8b0 size=0a17ch ( 41340) load[0m
[0;32mI (476) esp_image: segment 4: paddr=001f01ac vaddr=40819a30 size=03bf8h ( 15352) load[0m
[0;32mI (481) esp_image: segment 5: paddr=001f3dac vaddr=50000000 size=00068h ( 104) load[0m
[0;32mI (488) boot: Loaded app from partition at offset 0x60000[0m
[0;32mI (488) boot: Disabling RNG early entropy source...[0m
no external 32k oscillator, disable it now.
at param mode: 1
ESP-ROM:esp32c6-20220919
Build:Sep 19 2022
rst:0x8 (TG1_WDT_HPSYS),boot:0xc (SPI_FAST_FLASH_BOOT)
Saved PC:0x40800082
SPIWP:0xee
mode:DIO, clock div:1
load:0x4086c410,len:0xd2c
ets_loader.c 68
ESP-ROM:esp32c6-20220919
Build:Sep 19 2022
rst:0x7 (TG0_WDT_HPSYS),boot:0xc (SPI_FAST_FLASH_BOOT)
Saved PC:0x4001aa16
SPIWP:0xee
mode:DIO, clock div:1
load:0x4086c410,len:0xd2c
ets_loader.c 68
........
ets_loader.c 68

staunchheart, 2024-10-09:
usr other module is OK .but why?
all is ESPC6-WROOM  
[image]

ustccw, 2024-10-10:
same to https://www.esp32.com/viewtopic.php?f=42&t=42070

closed 2025-03-17 by ustccw