espressif/esp-at #878: Unable to build esp-at for Classic Bluetooth with Serial Port Protocol (SPP)

chrisj7903, 2024-10-10:
### Answers checklist

- [X] I have read the documentation [ESP-AT Programming Guide](https://docs.espressif.com/projects/esp-at/en/latest/esp32/Compile_and_Develop/How_to_clone_project_and_compile_it.html) and the issue is not addressed there.
- [X] I have compiled the hello_world example [Get Started with ESP-IDF](https://docs.espressif.com/projects/esp-at/en/latest/esp32/Compile_and_Develop/How_to_clone_project_and_compile_it.html#step-1-get-started-with-esp-idf) and the issue is not addressed there.
- [X] I have updated my ESP-AT branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### ESP-AT version

v3.4.0.0-46-g7b9487c2

### Operating System used

Windows

### If you are using Windows, please specify command line type.

CMD

### What is the expected behavior?

By following the instructions at [Compile ESP-AT Project Locally](https://docs.espressif.com/projects/esp-at/en/latest/esp32/Compile_and_Develop/How_to_clone_project_and_compile_it.html)
and [How to Enable ESP-AT Classic Bluetooth](https://docs.espressif.com/projects/esp-at/en/release-v2.4.0.0/esp32/Compile_and_Develop/how_to_enable_at_classic_bluetooth.html)
I expect to successfully build esp-at firmware, with Classic Bluetooth enabled with SPP.

### What is the actual behavior?

The build fails if both "AT bt command support" and "AT bt SPP command support" are enabled using the menuconfig utility as describned in the instructions.
However, if I build without SPP selected and only "AT bt command support" selected, the build & flashing succeed and the hardware responds corectly to basic Classic Bluetooth commands. 

### Steps to reproduce

Following the instructions at [Compile ESP-AT Project Locally](https://docs.espressif.com/projects/esp-at/en/latest/esp32/Compile_and_Develop/How_to_clone_project_and_compile_it.html)

cd %userprofile%\esp
git clone --recursive https://github.com/espressif/esp-at.git
cd esp-at
python build.py install
	choose 1. Platform_ESP32 
	choose 1. WROOM-32
	choose 1: Yes (silence mode enabled, to reduce firmware size)
python build.py menuconfig
	Component config
		AT -> AT bt command support (selected)
				-> AT bt SPP command support (selected)
		Quit with Save = Yes
python build.py build

the build fails with a fatal error as shown in logs following and attached iun ZIP file
[build_logs.zip](https://github.com/user-attachments/files/17326253/build_logs.zip)

### Build or installation Logs

```plain
Platform name:ESP32     Module name:WROOM-32    Silence:1
submodules check completed for updates.
esp_tls_parse_ecc_key.patch does not need to be applied.
fatfs_generation.patch does not need to be applied.
support_ext_partition.patch does not need to be applied.
modify_bluedroid_conn_timeout.patch does not need to be applied.
C:\Users\CJ\esp\esp-at\module_config\module_esp32_default\patch\libbtdm_app.a does not need to be applied.
patches check completed for updates.
Ready to set up environment variables..
PATH is C:\Espressif\tools\xtensa-esp-elf-gdb\14.2_20240403\xtensa-esp-elf-gdb\bin;C:\Espressif\tools\riscv32-esp-elf-gdb\14.2_20240403\riscv32-esp-elf-gdb\bin;C:\Espressif\tools\xtensa-esp-elf\esp-13.2.0_20230928\xtensa-esp-elf\bin;C:\Espressif\tools\riscv32-esp-elf\esp-13.2.0_20230928\riscv32-esp-elf\bin;C:\Espressif\tools\esp32ulp-elf\2.35_20220830\esp32ulp-elf\bin;C:\Espressif\tools\cmake\3.24.0\bin;C:\Espressif\tools\openocd-esp32\v0.12.0-esp32-20240318\openocd-esp32\bin;C:\Espressif\tools\ninja\1.11.1\;C:\Espressif\tools\idf-exe\1.0.3\;C:\Espressif\tools\ccache\4.8\ccache-4.8-windows-x86_64;C:\Espressif\tools\dfu-util\0.11\dfu-util-0.11-win64;C:\Espressif\frameworks\esp-idf-v5.2.2\tools;C:\Espressif\python_env\idf5.2_py3.11_env\Scripts\;C:\Espressif\tools\idf-git\2.44.0\cmd\;C:\Espressif;C:\Program Files (x86)\Common Files\Oracle\Java\javapath;C:\ProgramData\Oracle\Java\javapath;C:\Program Files (x86)\Intel\iCLS Client\;C:\Program Files\Intel\iCLS Client\;C:\WINDOWS\system32;C:\WINDOWS;C:\WINDOWS\System32\Wbem;C:\WINDOWS\System32\WindowsPowerShell\v1.0\;C:\Program Files (x86)\Intel\Intel(R) Management Engine Components\DAL;C:\Program Files\Intel\Intel(R) Management Engine Components\DAL;C:\Program Files (x86)\Intel\Intel(R) Management Engine Components\IPT;C:\Program Files\Intel\Intel(R) Management Engine Components\IPT;C:\WINDOWS\System32\OpenSSH\;C:\Program Files\MATLAB\R2024a\bin;C:\Program Files\MATLAB\R2023b\bin;C:\PROGRA~1\MATLAB\R2020b\bin;C:\Program Files (x86)\IVI Foundation\IVI\bin;C:\Program Files\IVI Foundation\IVI\bin;C:\Program Files (x86)\IVI Foundation\VISA\WinNT\Bin\;C:\Program Files\IVI Foundation\VISA\Win64\Bin\;C:\Program Files (x86)\IVI Foundation\VISA\WinNT\Bin;C:\Users\CJ\AppData\Local\Microsoft\WindowsApps;C:\Users\CJ\AppData\Local\Programs\MiKTeX 2.9\miktex\bin\x64\;
IDF_PYTHON_ENV_PATH is C:\Espressif\python_env\idf5.2_py3.11_env
sys.platform is win32
Not using an unsupported version of tool xtensa-esp-elf-gdb found in PATH: 14.2_20240403. To use it, run 'C:\Espressif\python_env\idf5.2_py3.11_env\Scripts\python.exe C:\Users\CJ\esp\esp-at\esp-idf\tools\idf_tools.py export --prefer-system'Not using an unsupported version of tool riscv32-esp-elf-gdb found in PATH: 14.2_20240403. To use it, run 'C:\Espressif\python_env\idf5.2_py3.11_env\Scripts\python.exe C:\Users\CJ\esp\esp-at\esp-idf\tools\idf_tools.py export --prefer-system'
Not using an unsupported version of tool xtensa-esp32-elf found in PATH: esp-13.2.0_20230928-13.2.0. To use it, run 'C:\Espressif\python_env\idf5.2_py3.11_env\Scripts\python.exe C:\Users\CJ\esp\esp-at\esp-idf\tools\idf_tools.py export --prefer-system'
Not using an unsupported version of tool xtensa-esp32s2-elf found in PATH: esp-13.2.0_20230928-13.2.0. To use it, run 'C:\Espressif\python_env\idf5.2_py3.11_env\Scripts\python.exe C:\Users\CJ\esp\esp-at\esp-idf\tools\idf_tools.py export --prefer-system'
Not using an unsupported version of tool xtensa-esp32s3-elf found in PATH: esp-13.2.0_20230928-13.2.0. To use it, run 'C:\Espressif\python_env\idf5.2_py3.11_env\Scripts\python.exe C:\Users\CJ\esp\esp-at\esp-idf\tools\idf_tools.py export --prefer-system'
[... 118 lines cut ...]
[1199/1201] Generating binary image from built executableesptool.py v4.7.0
Creating esp32 image...
Merged 2 ELF sections
Successfully created esp32 image.
Generated C:/Users/CJ/esp/esp-at/build/esp-at.bin
[1200/1201] cmd.exe /C "cd /D C:\Users\CJ\esp\esp-at\build...rtition-table.bin C:/Users/CJ/esp/esp-at/build/esp-at.bin"FAILED: esp-idf/esptool_py/CMakeFiles/app_check_size C:/Users/CJ/esp/esp-at/build/esp-idf/esptool_py/CMakeFiles/app_check_size
cmd.exe /C "cd /D C:\Users\CJ\esp\esp-at\build\esp-idf\esptool_py && C:\Espressif\python_env\idf5.2_py3.11_env\Scripts\python.exe C:/Users/CJ/esp/esp-at/esp-idf/components/partition_table/check_sizes.py --offset 0x8000 partition --type app C:/Users/CJ/esp/esp-at/build/partition_table/partition-table.bin C:/Users/CJ/esp/esp-at/build/esp-at.bin"
Error: All app partitions are too small for binary esp-at.bin size 0x183a00:
  - Part 'ota_0' 0/16 @ 0x100000 size 0x180000 (overflow 0x3a00)
  - Part 'ota_1' 0/17 @ 0x280000 size 0x180000 (overflow 0x3a00)
ninja: build stopped: subcommand failed.
ninja failed with exit code 1, output of the command is in the C:\Users\CJ\esp\esp-at\build\log\idf_py_stderr_output_4600 and C:\Users\CJ\esp\esp-at\build\log\idf_py_stdout_output_4600
A fatal error occurred: idf.py build failed
```

### More Information

I suspect some years back this was possible (to add Classic Bluetooth with SPP to ESP32 firmware). 
For example see this [description (circa 2021)](https://docs.ineltek.com/docs/bluetooth-spp-esp-at/)

However after many weeks of research I fear it may no longer possible as building with SPP always results in an oversized image/firmware size problem, no matter how many tweaks I tried (after studying [Minimizing Binary Size](https://docs.espressif.com/projects/esp-idf/en/stable/esp32/api-guides/performance/size.html))

There are just too many options and tweaks I don't understand, so I'm hoping either for guidance on the correct tweaks, or if necessary, an update to the esp-at menuconfig utiulity with new build options to support BT+SPP+A2DP as was originally intended. 

In my perfect world view: there would be a menuconfig option to build a completely Classic Bluetooth focussed, stripped down firmware (e.g no BLE, no WiFi at all). Hopefully this can could be done within the actual firmware size limits and constraints.

chrisj7903, 2024-10-10:
Supporting information - the sequence and output from idf.py size:

cd C:\Espressif\frameworks\esp-idf-v5.2.2\
install
cd %userprofile%\esp\esp-at
idf.py size
	Executing action: size
	Running ninja in directory C:\Users\...\esp\esp-at\build
	Executing "ninja all"...
	[1/1] cmd.exe /C "cd /D C:\Users\...\esp\esp-at\build\bootloader\esp-idf\es... bootloader 0x1000 C:/Users/.../esp/esp-at/build/bootloader/bootloader.bin"
	Bootloader binary size 0x6330 bytes. 0xcd0 bytes (11%) free.
	[3/5] Running utility command for customized_bin

	Creating NVS binary with version: V2 - Multipage Blob Support Enabled

	Created NVS binary: ===> C:/Users/.../esp/esp-at/build/customized_partitions\mfg_nvs.bin
	[4/5] cmd.exe /C "cd /D C:\Users\...\esp\esp-at\build\esp-idf\esptool_py &&...rtition_table/partition-table.bin C:/Users/.../esp/esp-at/build/esp-at.bin"
	FAILED: esp-idf/esptool_py/CMakeFiles/app_check_size C:/Users/.../esp/esp-at/build/esp-idf/esptool_py/CMakeFiles/app_check_size
	cmd.exe /C "cd /D C:\Users\...\esp\esp-at\build\esp-idf\esptool_py && C:\Espressif\python_env\idf5.2_py3.11_env\Scripts\python.exe C:/Users/.../esp/esp-at/esp-idf/components/partition_table/check_sizes.py --offset 0x8000 partition --type app C:/Users/.../esp/esp-at/build/partition_table/partition-table.bin C:/Users/.../esp/esp-at/build/esp-at.bin"
	Error: All app partitions are too small for binary esp-at.bin size 0x183a00:
	  - Part 'ota_0' 0/16 @ 0x100000 size 0x180000 (overflow 0x3a00)
	  - Part 'ota_1' 0/17 @ 0x280000 size 0x180000 (overflow 0x3a00)
	ninja: build stopped: subcommand failed.
	Running ninja in directory C:\Users\...\esp\esp-at\build
	Executing "ninja size"...
	[1/1] cmd.exe /C "cd /D C:\Users\...\esp\esp-at\build && C:\Espressif\python_env\idf5.2_py3.11_env\Scripts\python.exe C:/Users/.../esp/esp-at/esp-idf/tools/idf_size.py C:/Users/.../esp/esp-at/build/esp-at.map"
	Total sizes:
	Used static DRAM:   56848 bytes (  67732 remain, 45.6% used)
		  .data size:   20832 bytes
		  .bss  size:   36016 bytes
	Used static IRAM:  130490 bytes (    582 remain, 99.6% used)
		  .text size:  129463 bytes
	   .vectors size:    1027 bytes
	Used Flash size : 1436155 bytes
		  .text     : 1325311 bytes
		  .rodata   :  110588 bytes
Total image size: 1587477 bytes (.bin may be padded larger)

Luxin812, 2024-10-11:
"Error: All app partitions are too small for binary esp-at.bin size 0x183a00:"----> From this, it can be seen that the size of the esp-at.bin you compiled exceeds the current app partition. You can disable the AT wifi and mqtt functions that you do not use in menuconfig and try compiling again.

chrisj7903, 2024-10-11:
That makes sense, but can you please specify exactly which options I need to de-select in menuconfig to disable AT wifi & mqtt ? I have tried this idea before I have no expertise in menuconfig and there are many choices. So nothing I tried fixed my issue.

Luxin812, 2024-10-11:
Find the relevant configuration in menuconfig according to the following path and disable it.

Component config -> AT -> AT wifi command support
Component config -> AT -> AT smartconfig command support
Component config -> AT -> AT WPS command support
Component config -> AT -> AT MQTT command support

chrisj7903, 2024-10-13:
Two steps forward, one step back ... with WPS, smartconfig, WiFi & MQTT deselected, and with "bt command support" + "bt SPP command support" selected, I successfully completed the build, and then flashed to my ESP32-WROOM-32E.

However it but continually loops, starting up briefly then aborts and reboots every couple of seconds. I have attached the looping output below, with abort and backtrace information. If I repeat all the above but leave "bt SPP command support" de-selected I get same result, builds and flashes OK but continually reboots.

So the bt SPP capability remains ellusive, and I have no idea what to try next ... help!  

=============================================
Rebooting...
ets Jul 29 2019 12:21:46

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:5916
ho 0 tail 12 room 4
load:0x40078000,len:15844
load:0x40080400,len:3548
entry 0x40080600
[0;32mI (29) boot: ESP-IDF v5.0.6-dirty 2nd stage bootloader[0m
[0;32mI (29) boot: compile time 20:04:18[0m
[0;33mW (29) boot: Unicore bootloader[0m
[0;32mI (32) boot: chip revision: v3.0[0m
[0;32mI (36) boot.esp32: SPI Speed      : 40MHz[0m
[0;32mI (41) boot.esp32: SPI Mode       : DIO[0m
[0;32mI (45) boot.esp32: SPI Flash Size : 4MB[0m
[0;32mI (50) boot: Enabling RNG early entropy source...[0m
[0;32mI (55) boot: Partition Table:[0m
[0;32mI (59) boot: ## Label            Usage          Type ST Offset   Length[0m
[0;32mI (66) boot:  0 phy_init         RF data          01 01 0000f000 00001000[0m
[0;32mI (74) boot:  1 otadata          OTA data         01 00 00010000 00002000[0m
[0;32mI (81) boot:  2 nvs              WiFi data        01 02 00012000 0000e000[0m
[0;32mI (89) boot:  3 at_customize     unknown          40 00 00020000 000e0000[0m
[0;32mI (96) boot:  4 ota_0            OTA app          00 10 00100000 00180000[0m
[0;32mI (104) boot:  5 ota_1            OTA app          00 11 00280000 00180000[0m
[0;32mI (111) boot: End of partition table[0m
[0;32mI (115) esp_image: segment 0: paddr=00100020 vaddr=3f400020 size=185f4h ( 99828) map[0m
[0;32mI (160) esp_image: segment 1: paddr=0011861c vaddr=3ff80063 size=00008h (     8) load[0m
[0;32mI (160) esp_image: segment 2: paddr=0011862c vaddr=3ffbdb60 size=04ce0h ( 19680) load[0m
[0;32mI (174) esp_image: segment 3: paddr=0011d314 vaddr=40080000 size=02d04h ( 11524) load[0m
[0;32mI (179) esp_image: segment 4: paddr=00120020 vaddr=400d0020 size=10b778h (1095544) map[0m
[0;32mI (579) esp_image: segment 5: paddr=0022b7a0 vaddr=40082d04 size=19238h (102968) load[0m
[0;32mI (621) esp_image: segment 6: paddr=002449e0 vaddr=400c0000 size=00064h (   100) load[0m
[0;32mI (636) boot: Loaded app from partition at offset 0x100000[0m
[0;32mI (636) boot: Disabling RNG early entropy source...[0m
no external 32k oscillator, disable it now.

at param mode: 1

AT cmd port:uart1 tx:17 rx:16 cts:15 rts:14 baudrate:115200

module_name: WROOM-32

v3.4.0.0

abort() was called at PC 0x4016d696 on core 0

Backtrace: 0x40081626:0x3ffd30c0 0x4009373d:0x3ffd30e0 0x40097516:0x3ffd3100 0x4016d696:0x3ffd3170 0x4017d4b9:0x3ffd31a0 0x4017d542:0x3ffd31c0 0x4016d371:0x3ffd3210 0x400e88a6:0x3ffd3230 0x400ea957:0x3ffd3250 0x400eedf7:0x3ffd3280 0x401d8845:0x3ffd32b0

ELF file SHA256: 496f2f7abc0bca19

Luxin812, 2024-10-18:
Also turn off TCPIP commands. Maybe you can refer to my menuconfig configuration.
[image]
[image]

chrisj7903, 2024-10-18:
@luxin812 you are my hero, 
I just reflashed with your suggested AT settings and now I have same response to AT+BTSPPINIT=2 command as you did above
many, many thanks for your help :-)

closed 2025-03-17 by ustccw