espressif/esp-at #880: receive busy p... after restart

nimaltd, 2024-10-27:
### Answers checklist

- [X] I have read the documentation [ESP-AT Programming Guide](https://docs.espressif.com/projects/esp-at/en/latest/) and the issue is not addressed there.
- [X] I have used the latest released firmware or have updated my ESP-AT branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### AT+GMR

AT version:3.4.0.0(s-c31b833 - ESP32 - Jun  7 2024 03:48:17)
SDK version:v5.0.6-dirty
compile time(70ff5889):Jun  7 2024 04:46

### ESP-AT Firmware Source

1

### Hardware Information

esp32 wroom

### Power Supply used

USB

### What is the expected behavior?

working after restart

### What is the actual behavior?

at first power up, after connecting to a unavaliable SSID, and after restart, i receive busy p... for all command
 

### Probability of recurrence

after restart by key

### AT+SYSRAM?

+SYSRAM:162708,158344

### Steps to reproduce

step 1: all init and want connect to AP
step 2: after failed, i restart manually the esp32 by key
step 3: not possible to send new at

### AT command port output

```plain
ATC<ESP> - TX: ATE0

ATC<ESP> - RX: ATE0

ATC<ESP> - RX: OK

ATC<ESP> - TX: AT+SYSSTORE=0

ATC<ESP> - RX: OK

ATC<ESP> - TX: AT+GMR

ATC<ESP> - RX: AT version:3.4.0.0(s-c31b833 - ESP32 - Jun  7 2024 03:48:17)
SDK version:v5.0.6-dirty
compile time(70ff5889):Jun  7 2024 04:46
ATC<ESP> - TX: AT+SYSRAM?

ATC<ESP> - RX: +SYSRAM:162708,158344
OK

ATC<ESP> - TX: AT+CWMODE=1,0

ATC<ESP> - RX: OK

ATC<ESP> - TX: AT+CWJAP="Nima mobile","nima0000",,,,,,15

ATC<ESP> - RX: +CWJAP:3

ERROR
-------------------------------------------- restart by key and 

ATC<ESP> - TX: ATE0

ATC<ESP> - RX: 
busy p...
```

### AT log port output

```plain
no access
```

ustccw, 2024-10-29:
@nimaltd Hello, from the last few logs, it seems that after restarting, the `ATE0` command was sent before the `ready` message was output, which is usually inappropriate. Ideally, `ATE0` should be sent after the ESP32 restarts and the MCU receives the [ready](https://docs.espressif.com/projects/esp-at/en/latest/esp32/AT_Command_Set/index.html#id4) message. Please give this a try.

nimaltd, 2024-10-31:
Hello. Thanks. I will check

closed 2025-03-17 by ustccw