espressif/esp-at #884: .jpg file downloads with wrong header and file size

rsoric, 2024-11-06:
### Answers checklist.

- [X] I have read the documentation [ESP-AT Programming Guide](https://docs.espressif.com/projects/esp-at/en/latest/) and the issue is not addressed there.
- [X] I have used the latest released firmware or have updated my ESP-AT branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

Greetings, I have this issue with ESP-AT on ESP32 model ESP32C3, AT version:3.4.0.0-dev, SDK version:v5.0.6.

We are driving the ESP32C3 by an STM32 via SPI communication which we wrote our own driver for. Things generally work fine and we can download files. There is a problem with .jpg files which are hosted on GitHub though. For example, when downloading [this](https://gssc.esa.int/navipedia/images/a/a9/Example.jpg) image, the image downloads, decoded with our library and display on our device without problem. The bytes length of the downloaded file as printed in a debug mode matches the bytes length on my PC (61136 bytes).

However, when attempting to download [this](https://raw.githubusercontent.com/SolderedElectronics/Inkplate-Arduino-library/refs/heads/master/examples/Inkplate6/Advanced/WEB_WiFi/Inkplate6_Show_JPG_With_HTTPClient/image.jpg) image, the file size and contents don't match. On PC, the bytes size is 330040 bytes, and when downloaded via AT commands (in the same exact way as the other image) it's 329212 and our jpeg decoder can't decode the file and display it. Also the content-length header on ESP32 tells the byte size on PC (330040) but less bytes are received.

I've attempted to use a hex editor and a few days ago I was able to spot a couple different bytes in the header between the two files, but now I've just tried to recreate it to attach a screenshot to this issue and the headers of the file look exactly the same. I even tried going a couple thousand bytes deep and the files look the same. I compared the end of the file too and it's identical. So it's somewhere in the middle of the image data. I should also add that downloading files larger than the image works just fine and it retains the integrity of the file OK.

We have also tried to add headers `Accept-Encoding: identity`, `Connection: close` and `Content-Type: image/jpeg` - none of which worked. We thought it was a gzip issue and we had to ask the server for a non-gzipped file but this didn't work.

Any ideas as to what could be causing this?

BornaBiro, 2025-01-07:
Any update about this issue would be useful.
Do you guys from Espressif need more info about this issue (ex. logs)?
@ustccw could you help us out here? Or even connect with someone who can check this issue out?

Thanks.

ustccw, 2025-03-17:
sorry for a long delay. Could you please try the latest master code, and enable AT logs according to the following option, and see what happened.

```
Component config > Log output > Info （Or above Info）
Component config > AT > Enable ESP-AT Debug
Component config > AT > Enable ESP-AT Debug > Logging the data sent from AT to MCU (AT ---> MCU)
Component config > AT > Enable ESP-AT Debug > Logging the data sent from AT to MCU (AT ---> MCU) > The maximum length of the data sent from AT to MCU to be logged > 8192
Component config > AT > Enable ESP-AT Debug > Logging the data received by AT from MCU (AT <---- MCU)
Component config > AT > Enable ESP-AT Debug > Logging the data received by AT from MCU (AT <---- MCU) > The maximum length of the data received by AT from MCU to be logged > 8192
Component config > AT > Enable ESP-AT Debug > Enable Network Debug
```