espressif/esp-at #886: AT+BLESCAN retains previous filter information

JHedgesBartec, 2024-12-24:
### Answers checklist

- [X] I have read the documentation [ESP-AT Programming Guide](https://docs.espressif.com/projects/esp-at/en/latest/) and the issue is not addressed there.
- [X] I have used the latest released firmware or have updated my ESP-AT branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### AT+GMR

AT+GMR
AT version:3.5.0.0-dev(752008f - ESP32C6 - Dec 12 2024 12:47:58)
SDK version:v5.1.5-196-g64849cb703-dirty
compile time(24070dcd):Dec 24 2024 09:27:20
Bin version:v4.1.0.0-dev(ESP32C6-4MB)

OK

### ESP-AT Firmware Source

Built from Master 24/12/2024 as release v4.0.0.0 has a bug where scan filtering is inoperable.

### Hardware Information

ESP32-C6

### Power Supply used

USB

### What is the expected behavior?

When sending the AT+BLESCAN command, filtering should only be done 

### What is the actual behavior?

If the user sends a command to filter BLE SCAN results, for example by name, then starts a different scan with no filtering, then the scan results are filtered as per the first scan.

### Probability of recurrence

100%

### AT+SYSRAM?

AT+SYSRAM?
+SYSRAM:232696,223916
OK

### Steps to reproduce

1. Initialise BLE as Client
2. Configure scan parameters (or leave as default)
3. Start Scan with a NAME filter
4. Only devices with that name are shown
5. Start Scan with **no** filter at all
6. Only devices with the name from the previous scan are shown
7. Reboot module (using AT+BLEINIT=0 causes a reboot anyway)
8. Initialise BLE as Client
9. Configure scan parameters (or leave as default)
10. Start Scan with **no** filter at all
11. All devices are shown

### AT command port output

```plain
(09:35:47.201) ATE0
(09:35:47.268) ATE0
(09:35:47.268) 
(09:35:47.268) OK
(09:35:48.429) AT+GMR
(09:35:48.493) AT version:3.5.0.0-dev(752008f - ESP32C6 - Dec 12 2024 12:47:58)
(09:35:48.493) SDK version:v5.1.5-196-g64849cb703-dirty
[... 171 lines cut ...]
(09:36:13.006) +BLESCAN:"44:c0:ff:88:f8:b0",-81,,14ff4c000100000000000000000000000080000000,1,4
(09:36:13.006) +BLESCAN:"1e:f9:43:ba:2a:a5",-63,1eff0600010920227438f59062039a823cf78c7cdbe7122944fc845168f8a6,,1,3
(09:36:13.006) +BLESCAN:"08:b6:1f:9b:d3:12",-51,02010603031dba1009543630305f35353031383339333730,,0,0
(09:36:13.006) +BLESCAN:"3a:6e:51:cf:98:bc",-72,1eff060001092002b7bbfd7b35195d85b3c360e475c4f52843c407cc2f9ba9,,1,3
(09:36:13.006) +BLESCAN:"19:44:9e:b1:68:85",-81,0201021816f1fc042fe2f98f2c0649b052b7dea401a980c9f45a5421,,1,0
(09:36:13.006) +BLESCAN:"7d:9d:55:86:8e:d4",-73,02011a1aff4c00021554d9faadb2b64193a835c899e8ed0d8300010001c5,,1,0
(09:36:13.006) +BLESCAN:"7d:9d:55:86:8e:d4",-73,,,1,4
(09:36:13.006) +BLESCAN:"1d:82:70:e4:4a:40",-93,1eff0600010920022b02d89055a3358a770e053d42b439a65db7f0ca7b3640,,1,3
(09:36:13.006) +BLESCAN:"15:e0:ec:37:79:03",-85,0201021916f1fc04550d4048fb83a033b664fcb8e6b3b99b691399744a,,1,0
(09:36:13.006) +BLESCAN:"15:e0:ec:37:79:03",-85,,,1,4
(09:36:13.006) +BLESCAN:"26:77:4e:54:5a:30",-54,1eff060001092022143f3f3ff831fa0d71b11485b50f8bb687552676ed4467,,1,3
(09:36:13.006) +BLESCAN:"66:cd:72:29:94:6a",-66,1eff06000109200245836764f8e1a814cfa378484ea8a4ca6c883eb49c52d2,,1,3
(09:36:13.006) +BLESCAN:"08:ac:c6:ef:3c:d0",-83,1e16f3fe4a17234c3048491132d529e737f1a50159d6efcf661477743adb51,,1,3
(09:36:13.006) +BLESCANDONE
```

### AT log port output

```plain
Output too large, nothing shown when starting a scan
```

### More Information.

None

JHedgesBartec, 2024-12-24:
@ustccw @Luxin812 is it possible for this to be fixed ahead of the planned January/February release?

Luxin812, 2025-05-21:
@JHedgesBartec I've reproduced the issue, and it will be fixed and merged into the master branch soon. Thank you for bringing this to our attention!

closed 2025-07-30 by igrr

ustccw, 2025-07-30:
@JHedgesBartec Sorry that this issue was unintentionally overlooked for a very long time. It has now been fixed on the master branch. If you have a chance to revisit it, we'd really appreciate your help in verifying the fix.

ustccw, 2025-08-11:
@JHedgesBartec Thank you for your confirmation!