espressif/esp-at #887: AT+BLEINIT=0 causes a reboot (CORRUPT HEAP)

JHedgesBartec, 2024-12-24:
### Answers checklist

- [X] I have read the documentation [ESP-AT Programming Guide](https://docs.espressif.com/projects/esp-at/en/latest/) and the issue is not addressed there.
- [X] I have used the latest released firmware or have updated my ESP-AT branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### AT+GMR

AT+GMR
AT version:3.5.0.0-dev(752008f - ESP32C6 - Dec 12 2024 12:47:58)
SDK version:v5.1.5-196-g64849cb703-dirty
compile time(24070dcd):Dec 24 2024 09:27:20
Bin version:v4.1.0.0-dev(ESP32C6-4MB)

OK

### ESP-AT Firmware Source

Built from Master 24/12/2024

### Hardware Information

ESP32-C6

### Power Supply used

USB

### What is the expected behavior?

BLE deinits without rebooting and affecting other initialised radio (WiFi)

### What is the actual behavior?

Reboots when AT+BLEINIT=0 is sent, log shows CORRUPT_HEAP and call stack before reboot.

### Probability of recurrence

100%

### AT+SYSRAM?

AT+SYSRAM?
+SYSRAM:186724,186260
OK

### Steps to reproduce

1. Initialise BLE (as Client)
2. Deinit BLE

### AT command port output

```plain
ATE0
ATE0

OK
AT+GMR
AT version:3.5.0.0-dev(752008f - ESP32C6 - Dec 12 2024 12:47:58)
SDK version:v5.1.5-196-g64849cb703-dirty
compile time(24070dcd):Dec 24 2024 09:27:20
Bin version:v4.1.0.0-dev(ESP32C6-4MB)

OK
AT+BLEINIT=1

OK
AT+BLEINIT=0

[... 188 lines cut ...]
(09:45:44.987) <ESC>[0;32mI (64) boot: Enabling RNG early entropy source...<ESC>[0m
(09:45:44.987) <ESC>[0;32mI (70) boot: Partition Table:<ESC>[0m
(09:45:44.987) <ESC>[0;32mI (73) boot: ## Label            Usage          Type ST Offset   Length<ESC>[0m
(09:45:44.987) <ESC>[0;32mI (80) boot:  0 otadata          OTA data         01 00 0000d000 00002000<ESC>[0m
(09:45:45.010) <ESC>[0;32mI (88) boot:  1 phy_init         RF data          01 01 0000f000 00001000<ESC>[0m
(09:45:45.010) <ESC>[0;32mI (95) boot:  2 nvs              WiFi data        01 02 00010000 0000e000<ESC>[0m
(09:45:45.010) <ESC>[0;32mI (103) boot:  3 at_customize     unknown          40 00 0001e000 00042000<ESC>[0m
(09:45:45.037) <ESC>[0;32mI (110) boot:  4 ota_0            OTA app          00 10 00060000 001d0000<ESC>[0m
(09:45:45.037) <ESC>[0;32mI (118) boot:  5 ota_1            OTA app          00 11 00230000 001d0000<ESC>[0m
(09:45:45.037) <ESC>[0;32mI (126) boot: End of partition table<ESC>[0m
(09:45:45.037) <ESC>[0;32mI (130) esp_image: segment 0: paddr=00060020 vaddr=42160020 size=2b3a8h (177064) map<ESC>[0m
(09:45:45.113) <ESC>[0;32mI (210) esp_image: segment 1: paddr=0008b3d0 vaddr=40800000 size=04c48h ( 19528) load<ESC>[0m
(09:45:45.113) <ESC>[0;32mI (220) esp_image: segment 2: paddr=00090020 vaddr=42000020 size=151cbch (1383612) map<ESC>[0m
(09:45:45.670) <ESC>[0;32mI (776) esp_image: segment 3: paddr=001e1ce4 vaddr=40804c48 size=18a6ch (100972) load<ESC>[0m
(09:45:45.717) <ESC>[0;32mI (823) esp_image: segment 4: paddr=001fa758 vaddr=4081d6c0 size=0402ch ( 16428) load<ESC>[0m
(09:45:45.742) <ESC>[0;32mI (832) esp_image: segment 5: paddr=001fe78c vaddr=50000000 size=00068h (   104) load<ESC>[0m
(09:45:45.742) <ESC>[0;32mI (846) boot: Loaded app from partition at offset 0x60000<ESC>[0m
(09:45:45.747) <ESC>[0;32mI (847) boot: Disabling RNG early entropy source...<ESC>[0m
(09:45:46.186) no external 32k oscillator, disable it now.<CR>
(09:45:46.222) at param mode: 1<CR>
(09:45:46.281) AT cmd port:uart1 tx:7 rx:6 cts:5 rts:4 baudrate:115200<CR>
(09:45:46.281) module_name: ESP32C6-4MB<CR>
(09:45:46.310) max tx power=78, ret=0<CR>
(09:45:46.310) v4.1.0.0-dev (unknown)<CR>
```

JHedgesBartec, 2024-12-24:
@ustccw @Luxin812 is it possible for this to be fixed ahead of the planned January/February release?

ustccw, 2024-12-25:
yes. we are testing new version for esp32,esp32c2,esp32c3,esp32c6. it should get fixed in new version.

ustccw, 2025-03-17:
@JHedgesBartec This issue should be fixed on the latest master branch.