espressif/esp-at #889: By default, the PHY mode of ESP8266 devices is 802.11bg mode, and the PHY mode of non-ESP8266 devices is 802.11bgn mode. Can ESP8266 support 802.11bgn by default by modifying the code?

aithinker2222, 2025-01-06:
### Answers checklist.

- [X] I have read the documentation [ESP-AT Programming Guide](https://docs.espressif.com/projects/esp-at/en/latest/) and the issue is not addressed there.
- [X] I have used the latest released firmware or have updated my ESP-AT branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

Set Command
^^^^^^^^^^^

**Command:**

::

    AT+CWSTAPROTO=<protocol>

**Response:**

::

    OK

Parameters
^^^^^^^^^^

-  **<protocol>**:

   -  bit0: 802.11b protocol standard.
   -  bit1: 802.11g protocol standard.
   -  bit2: 802.11n protocol standard.

Note
^^^^^

-  Currently ESP devices only support 802.11b or 802.11bg or 802.11bgn mode.
-  By default, PHY mode of ESP8266 device is 802.11bg mode, and PHY mode of non ESP8266 device is 802.11bgn mode.

aithinker2222, 2025-01-06:
I need ESP8266 to support 802.11bgn mode by default without sending the AT+CWSTAPROTO command. How should I modify the code?

ustccw, 2025-03-17:
@aithinker2222 Hi! you can call:
```
esp_wifi_set_protocol(ESP_IF_WIFI_STA, WIFI_PROTOCOL_11B | WIFI_PROTOCOL_11G | WIFI_PROTOCOL_11N);
```

after [at_custom_init()](https://github.com/espressif/esp-at/blob/release/v2.2.0.0_esp8266/main/app_main.c#L278)