espressif/esp-at #890: Unable to accept +BLESECNCREQ if initiating a connection where the server still has a paring but the client does not

JHedgesBartec, 2025-01-06:
### Answers checklist

- [X] I have read the documentation [ESP-AT Programming Guide](https://docs.espressif.com/projects/esp-at/en/latest/) and the issue is not addressed there.
- [X] I have used the latest released firmware or have updated my ESP-AT branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### AT+GMR

AT+GMR
AT version:3.5.0.0-dev(752008f - ESP32C6 - Dec 12 2024 12:47:58)
SDK version:v5.1.5-196-g64849cb703-dirty
compile time(a107039a):Jan  6 2025 09:01:51
Bin version:v1.0.1.0(ESP32C6-4MB)

### ESP-AT Firmware Source

Built from master.

### Hardware Information

ESP32-C6 WROOM

### Power Supply used

External 3.3V

### What is the expected behavior?

If a pairing has been removed from the client but not the server, if the client imitates a connection upon reception of `+BLESECNCREQ:<connection index>,<key>` the client should be able to respond with `+BLECONFREPLY`

### What is the actual behavior?

When the client sends the `+BLESECNCREQ` response over the uart, if `+BLECONFREPLY` is sent the module responds with `busy p...` as the connection command is in progress still

### Probability of recurrence

100%

### AT+SYSRAM?

AT+SYSRAM?
+SYSRAM:232108,224356
OK

### Steps to reproduce

1. Initialise the ESP32 as a BLE client
2. Create a pairing to a server
3. Delete the pairing only on the client
4. Attempt to connect to the server
5. `+BLESECNCREQ` is received
6. Send `+BLECONFREPLY`
7. Module responds with `busy p...` and fails the connection

### AT command port output

```plain
ATE0
ATE0

OK
AT+GMR
AT version:3.5.0.0-dev(752008f - ESP32C6 - Dec 12 2024 12:47:58)
SDK version:v5.1.5-196-g64849cb703-dirty
compile time(a107039a):Jan  6 2025 09:01:51
Bin version:v1.0.1.0(ESP32C6-4MB)

OK
AT+CWINIT=0

OK
AT+BLEINIT=1

OK
AT+BLESCANPARAM=1,0,0,100,50

OK
AT+BLESCAN=1,1

OK
+BLESCAN:"42:d1:9c:42:73:28",-56,,0d094241522d4f462d5453573134,1,4
+BLESCANDONE
AT+BLECONN=0,"42:d1:9c:42:73:28",1
+BLECONN:0,"42:d1:9c:42:73:28"
+BLECFGMTU:0,203
+BLECONNPARAM:0,12,12,12,0,960
BLE_TT_MODE

OK
+BLECONNPARAM:0,48,48,48,0,960
AT+BLEENC=0,3

OK
+BLESECNCREQ:0,714177
AT+BLECONFREPLY=0,1

OK
+BLEAUTHCMPL:0,0
+READ:0,"42:d1:9c:42:73:28"
+BLECONNPARAM:0,12,12,12,0,960
+READ:0,"42:d1:9c:42:73:28"
+READ:0,"42:d1:9c:42:73:28"
+READ:0,"42:d1:9c:42:73:28"
+READ:0,"42:d1:9c:42:73:28"
+BLECONNPARAM:0,48,48,48,0,960
AT+BLEDISCONN=0

OK
+BLEDISCONN:0,"42:d1:9c:42:73:28"
AT+BLEENCDEV?
+BLEENCDEV:0,"42:d1:9c:42:73:28"

OK
AT+BLEENCCLEAR=0

OK
AT+BLESCAN=1,1

OK
+BLESCAN:"42:d1:9c:42:73:28",-52,,0d094241522d4f462d5453573134,1,4
+BLESCANDONE
AT+BLECONN=0,"42:d1:9c:42:73:28",1
+BLECONN:0,"42:d1:9c:42:73:28"
+BLESECNCREQ:0,659254
AT+BLECONFREPLY=0,1

busy p...
BLE_TT_MODE

OK
+BLEAUTHCMPL:0,1
+BLEDISCONN:0,"42:d1:9c:42:73:28"
AT+SYSRAM?
+SYSRAM:232108,224356
OK
```

### AT log port output

```plain
connection established; status=0 handle=0 our_ota_addr_type=0 our_ota_addr=4083bc9f:00:15:4082c000:00:42176bc8
 our_id_addr_type=0 our_id_addr=4083bc91:00:15:4082c000:00:42176bc8
 peer_ota_addr_type=1 peer_ota_addr=4083bca6:00:15:4082c000:00:42176bc8
 peer_id_addr_type=1 peer_id_addr=4083bc98:00:15:4082c000:00:42176bc8
 conn_itvl=40 conn_latency=0 supervision_timeout=256 encrypted=0 authenticated=0 bonded=0
PASSKEY_ACTION_EVENT started 
encryption change event; status=7 handle=0 our_ota_addr_type=0 our_ota_addr=4083bbcf:00:15:4082c000:00:42176bc8
 our_id_addr_type=0 our_id_addr=4083bbc1:00:15:4082c000:00:42176bc8
 peer_ota_addr_type=1 peer_ota_addr=4083bbd6:00:15:4082c000:00:42176bc8
 peer_id_addr_type=1 peer_id_addr=4083bbc8:00:15:4082c000:00:42176bc8
 conn_itvl=40 conn_latency=0 supervision_timeout=256 encrypted=0 authenticated=0 bonded=0

disconnect; reason=534 handle=0 our_ota_addr_type=0 our_ota_addr=4083bd1b:00:15:4082c000:00:42176bc8
 our_id_addr_type=0 our_id_addr=4083bd0d:00:15:4082c000:00:42176bc8
 peer_ota_addr_type=1 peer_ota_addr=4083bd22:00:15:4082c000:00:42176bc8
 peer_id_addr_type=1 peer_id_addr=4083bd14:00:15:4082c000:00:42176bc8
 conn_itvl=40 conn_latency=0 supervision_timeout=256 encrypted=0 authenticated=0 bonded=0
```

Luxin812, 2025-05-21:
@JHedgesBartec The AT+BLECONN command is blocking. Please ensure that you wait for this command to complete (i.e., until you receive OK or ERROR) before executing the next AT command.

Luxin812, 2025-09-09:
Your needs have been understood, and we will conduct an internal evaluation to determine the best solution.

XionHsieh1025, 2025-12-04:
The situation you mentioned does exist. 

On the client side, after the connection is successfully established, service discovery will be performed automatically. Only when the service discovery is completed will an "OK" prompt be reported, indicating that the connection has been established successfully.

Before reporting the "OK" prompt, some parameter interactions will occur. These include updates to connection parameters, MTU updates, and SMP-related prompts, as these processes can be interleaved during service discovery.

Therefore, if the main control MCU intends to send the corresponding response command, it still needs to wait for the "OK" response to be returned.