espressif/esp-at #895: Tx in pin 35 doesn't work

pedrorovi, 2025-01-21:
### Answers checklist

- [x] I have read the documentation [ESP-AT Programming Guide](https://docs.espressif.com/projects/esp-at/en/latest/) and the issue is not addressed there.
- [x] I have used the latest released firmware or have updated my ESP-AT branch (master or release) to the latest version and checked that the issue is present there.
- [x] I have searched the issue tracker for a similar issue and not found a similar issue.

### AT+GMR

AT+GMR
AT version:4.0.0.0-dev(s-3fe3806 - ESP32 - Dec 29 2023 11:11:33)
SDK version:v5.0.4-dirty
compile time(81b671ef):Jan 21 2025 07:59:13
Bin version:v4.0.0.0(WROOM-32)

OK

### ESP-AT Firmware Source

3 - Compile ESP-AT project locally.

### Hardware Information

ESP32 Wroom Olimex Gateway

### Power Supply used

USB

### What is the expected behavior?

I have been used the pin 35 as TX for the console, and now it doesn't work. Instead, it works in pin 2, but we actually have many devices with the hardware design.
We use the release 2.4 and 3.2, and this changes when we try to update it to 4.0.

### What is the actual behavior?

Console doesn't work.

### Probability of recurrence

Always

### AT+SYSRAM?

AT+SYSRAM?
+SYSRAM:153480,149040
OK

### Steps to reproduce

1. Change uart1 tx pin in factory param data csv.
2. Enable ethernet.
3. Compile.
4. Flash

### AT command port output

```plain
Can't see anything. The examples that I use in the other questions is using pin 2.
```

### AT log port output

```plain
ets Jul 29 2019 12:21:46

rst:0x1 (POWERON_RESET),boot:0x1a (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:5916
ho 0 tail 12 room 4
load:0x40078000,len:15184
load:0x40080400,len:3540
entry 0x40080600
I (29) boot: ESP-IDF v5.0.4-dirty 2nd stage bootloader
I (29) boot: compile time 16:36:49
W (29) boot: Unicore bootloader
I (33) boot: chip revision: v3.0
I (37) boot.esp32: SPI Speed      : 40MHz
I (41) boot.esp32: SPI Mode       : DIO
I (46) boot.esp32: SPI Flash Size : 4MB
I (50) boot: Enabling RNG early entropy source...
I (56) boot: Partition Table:
I (59) boot: ## Label            Usage          Type ST Offset   Length
I (67) boot:  0 phy_init         RF data          01 01 0000f000 00001000
I (74) boot:  1 otadata          OTA data         01 00 00010000 00002000
I (81) boot:  2 nvs              WiFi data        01 02 00012000 0000e000
I (89) boot:  3 at_customize     unknown          40 00 00020000 000e0000
I (96) boot:  4 factory          factory app      00 00 00100000 00180000
I (104) boot:  5 spiffs           Unknown data     01 82 00280000 00180000
I (111) boot: End of partition table
I (116) boot: Defaulting to factory image
I (120) esp_image: segment 0: paddr=00100020 vaddr=3f400020 size=1aee4h (110308) map
I (169) esp_image: segment 1: paddr=0011af0c vaddr=3ff80063 size=00008h (     8) load
I (169) esp_image: segment 2: paddr=0011af1c vaddr=3ffbdb60 size=048fch ( 18684) load
I (182) esp_image: segment 3: paddr=0011f820 vaddr=40080000 size=007f8h (  2040) load
I (184) esp_image: segment 4: paddr=00120020 vaddr=400d0020 size=133314h (1258260) map
I (646) esp_image: segment 5: paddr=0025333c vaddr=400807f8 size=1f0b8h (127160) load
I (699) esp_image: segment 6: paddr=002723fc vaddr=400c0000 size=00064h (   100) load
I (715) boot: Loaded app from partition at offset 0x100000
I (715) boot: Disabling RNG early entropy source...
at param mode: 1
AT cmd port:uart1 tx:35 rx:16 cts:-1 rts:-1 baudrate:115200
module_name: WROOM-32
max tx power=78, ret=0
v4.0.0.0
```

ustccw, 2025-03-17:
@pedrorovi Hi, this hardware seem to be from [Olimex](https://www.olimex.com/Products/IoT/ESP32/ESP32-GATEWAY/open-source-hardware), not from [espressif](https://www.espressif.com/en/products/modules). I would like to suggest seek the help from Olimex or buy an espressif module for a test.

btw, from the log you provided, if you uses ESP32-WROOM series, there is no an ethernet interface. and you should change `PLATFORM_ESP32,WROOM-32,...` line.

pedrorovi, 2025-03-21:
Yes @ustccw , it is a Olimex board, but with older versions works using these pins. The ESP32-WROOM is the microcontroller that works on Olimex ESP32-Gateway -> [https://www.olimex.com/Products/IoT/ESP32/ESP32-GATEWAY/open-source-hardware](url)

ustccw, 2025-03-21:
From the https://www.olimex.com/Products/IoT/ESP32/ESP32-GATEWAY/resources/ESP32-GATEWAY-GPIOs-Rev.F-up.pdf, GPIO35 should be free to use. and the esp logs also don't output any wrong message. It's pretty weird. could you please try with the other GPIO.

The best approach is to contact someone from Olimex to help you confirm whether your software and hardware are compatible.