espressif/esp-at #896: AT+WSCFG returns error when trying to set authentication to any of the possible options

fahirad, 2025-01-22:
### Answers checklist.

- [x] I have read the documentation [ESP-AT Programming Guide](https://docs.espressif.com/projects/esp-at/en/latest/) and the issue is not addressed there.
- [x] I have used the latest released firmware or have updated my ESP-AT branch (master or release) to the latest version and checked that the issue is present there.
- [x] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

I have been trying to follow this example [https://docs.espressif.com/projects/esp-at/en/latest/esp32c3/AT_Command_Examples/websocket_at_examples.html#websocket-connection-over-tls-mutual-authentication](url) as my basis for websocket connection. My ESP32-C3 is a client in this scenario. However the command for configuring the websocket component returns error if I send any value for _auth_ field. I can't seem to find any issues similar to this one. 
Any hints would be greatly appreciated!

fahirad, 2025-01-22:
I was using the wrong version after all, my bad.

fahirad, 2025-01-28:
Soo I still do need help.
A bit of background, we are using forked esp-at repository with original one as the upstream remote in it, so that we can share our changes (mostly pin related) among ourselves and also access new releases. 
It is so troublesome to update it first of all. I have managed to do it but after a lot of money spent. 
Shouldn't somewhere be a guide regarding how to update the esp-at other than cloning it again?

Back to the mentioned command. On the documentation site, when the master branch is selected for ESP32-C3, command WSCFG has additional entries for authentication. However after I have updated it and flashed my esp device it still did not work. Release v4.0 which seems like the latest one also doesn't have it.
My question is: Is authentication with websockets possible on ESP32C3 and if yes which version do I need to have to obtain it?

ustccw, 2025-03-17:
@fahirad Hi, esp-at repository allows you to override all the souce code, intead of always cloning it. you can read this example:
https://github.com/espressif/esp-at/tree/master/examples/at_override_module_config
in your case, you can make a patch for your changes, and put it to [patch](https://github.com/espressif/esp-at/tree/master/examples/at_override_module_config/patch), specify it in [patch_list.ini](https://github.com/espressif/esp-at/blob/master/examples/at_override_module_config/patch/patch_list.ini).

WebSocket authentication is added at [18255f6b](https://github.com/espressif/esp-at/commit/18255f6bda1d0326cb85088682ab8562afe6fea4) (24/9/2024) for all chips. and v4.0 (29/12/2024) doesn't include this change. please know.