espressif/esp-at #899: AT+BLESCAN name filter fails if the name has a null character in it

JHedgesBartec, 2025-01-31:
### Answers checklist

- [x] I have read the documentation [ESP-AT Programming Guide](https://docs.espressif.com/projects/esp-at/en/latest/) and the issue is not addressed there.
- [x] I have used the latest released firmware or have updated my ESP-AT branch (master or release) to the latest version and checked that the issue is present there.
- [x] I have searched the issue tracker for a similar issue and not found a similar issue.

### AT+GMR

Official release firmware: v3.4.0.0 (https://docs.espressif.com/projects/esp-at/en/latest/esp32/AT_Binary_Lists/esp_at_binaries.html)

### ESP-AT Firmware Source

Official Release firmware.

### Hardware Information

ESP32 WROOM

### Power Supply used

USB

### What is the expected behavior?

When providing a name to filter scan results for, the comparison should be done based on string length of the advertised name.

### What is the actual behavior?

If the broadcast name in the scan response data contains '00' (trailling null as part of a string) the name filter is failing.
For example:
+BLESCAN:"f8:8a:5e:0d:14:c0",-36,,0302221108094553502D415400,0,0

Where the name field is 08094553502D415400 which is "ESP-AT" followed by a null. If the AT+BLESCAN command is provided "ESP-AT" no scan results are returned.

Whist having the null in the name may not be considered valid, some third party devices include it, so the ESP-AT framework should account for this.

### Probability of recurrence

100%

### AT+SYSRAM?

AT+SYSRAM?
+SYSRAM:118296,115376
OK

### Steps to reproduce

Configure a server to broadcast the following data in the name 4553502D415400 ("ESP-AT" followed by NULL)
Start a scan using "ESP-AT" as the name filter
No scan data is returned

### AT command port output

```plain
AT+BLESCAN=1,30,2,"ESP-AT"

OK
+BLESCANDONE
```

### AT log port output

```plain
Built with silence mode enabled.
```

JHedgesBartec, 2025-09-05:
Hi @XionHsieh1025,

Without seeing the source code for AT, I believe the problem is not parsing the parameters within the AT command but comparison of the parameter to the complete local name in the BLE ADV data.

If the comparison method is changed to substring match then this should be enough to work around this issue, though I would require a build to test to confirm this.

XionHsieh1025, 2026-04-16:
Thanks. This issue seems to only happen on ESP32.
I see you’re also using chips like ESP32-C6 — does the C6 have this issue too?