espressif/esp-at #903: What is the behaviour of AT+BLEGATTCWR when the server supports both 'Write' and 'Write without Response'

JHedgesBartec, 2025-02-18:
If a third party device (non-espressif) hosts a service and characteristic which has both 'Write' and 'Write without Response', when the AT+BLEGATTCWR command is used to write to the characteristic, which method is used?

Does the command prioritise Write?
Is there a way to enforce the 'Write' method is used?

JHedgesBartec, 2025-02-18:
@Luxin812 or @ustccw are you able to clarify this as it is not present in the documentation?

XionHsieh, 2025-02-28:
If  'Write' and 'Write without Response' are supported, ‘write’ is used by default.

ustccw, 2025-03-17:
@Luxin812 PTAL

Luxin812, 2025-03-25:
@JHedgesBartec Do you have any more questions about this issue? If not, maybe you could close it, thanks.

JHedgesBartec, 2025-03-25:
@Luxin812 are you confirming the behaviour that @XionHsieh has reported?
It is not clear who is an employee of Espressif and such has access to the source code to confirm this issue.

Luxin812, 2025-03-26:
@JHedgesBartec Theoretically, it should default to using `Write`. Perhaps you can verify this behavior by capturing packets. If the captured data does not show `Write`, we can discuss the issue further.

JHedgesBartec, 2025-03-26:
@Luxin812 how is it possible to capture the packets?

In this case we are using the ESP32 with BLE, BT Classic and WiFi, due to this we cannot enable logging on the build and we are unable to remove the unused the OTA partitioning to allow more flash space for the application.

Luxin812, 2025-09-09:
You can start by searching online for relevant BLE air packet capturing tutorials to better understand the operational steps.

closed 2026-04-03 by ustccw