espressif/esp-at #909: AT+BLECONNPARAM? returns all 0 for a connection in server mode

JHedgesBartec, 2025-02-19:
### Answers checklist

- [x] I have read the documentation [ESP-AT Programming Guide](https://docs.espressif.com/projects/esp-at/en/latest/) and the issue is not addressed there.
- [x] I have used the latest released firmware or have updated my ESP-AT branch (master or release) to the latest version and checked that the issue is present there.
- [x] I have searched the issue tracker for a similar issue and not found a similar issue.

### AT+GMR

AT+GMR
AT version:3.5.0.0-dev(8bfb16f - ESP32C6 - Jan  6 2025 09:10:51)
SDK version:v5.1.5-196-g64849cb703-dirty
compile time(5302cdc7):Jan 29 2025 11:35:15
Bin version:v1.0.1.1(ESP32C6-4MB)

OK

### ESP-AT Firmware Source

Compiled from master but with AT lib patched with a version provided by an Espressif employee to fix an issue with characteristics and pairings

### Hardware Information

ESP32-C6

### Power Supply used

External 3.3V

### What is the expected behavior?

When `AT+BLECONNPARAM?` is sent with an established connection the connection parameters should return non-zero

### What is the actual behavior?

The command returns with all 0's for the connection parameters despite the log showing they are set.

The `+BLECONNPARAM` is also not seen when the connection is formed, something that is seen with the ESP32 and Bluedroid running as BLE server.

I am also seeing reliability issues when sending data to a connected client, whereby the client does not receive the data, but the ESP32 server says send was OK, which lead me to query the connection parameters.

### Probability of recurrence

100%

### AT+SYSRAM?

AT+SYSRAM?
+SYSRAM:233736,223820
OK

### Steps to reproduce

1. Configure as a BLE server using the ESP32-C6
2. Using a third party application connect to the server
3. Send command to query connection parameters

### AT command port output

```plain
+BLECONN:0,"d4:54:8b:fd:6d:0c"
+BLEAUTHCMPL:0,0
+WRITE:2

+BLECFGMTU:0,203
AT+BLECONNPARAM?
+BLECONNPARAM:0,0,0,0,0,0

OK
```

### AT log port output

```plain
connection established; status=0 handle=0 our_ota_addr_type=0 our_ota_addr=4083afdf:00:15:4082c000:00:42176c70
 our_id_addr_type=0 our_id_addr=4083afd1:00:15:4082c000:00:42176c70
 peer_ota_addr_type=0 peer_ota_addr=4083afe6:00:15:4082c000:00:42176c70
 peer_id_addr_type=0 peer_id_addr=4083afd8:00:15:4082c000:00:42176c70
 conn_itvl=48 conn_latency=0 supervision_timeout=960 encrypted=0 authenticated=0 bonded=0
advertise complete; reason=0encryption change event; status=0 handle=0 our_ota_addr_type=0 our_ota_addr=4083af6f:00:15:4082c000:00:42176c70
 our_id_addr_type=0 our_id_addr=4083af61:00:15:4082c000:00:42176c70
 peer_ota_addr_type=0 peer_ota_addr=4083af76:00:15:4082c000:00:42176c70
 peer_id_addr_type=0 peer_id_addr=4083af68:00:15:4082c000:00:42176c70
 conn_itvl=48 conn_latency=0 supervision_timeout=960 encrypted=1 authenticated=1 bonded=1

subscribe event; conn_handle=0 attr_handle=8 reason=3 prevn=0 curn=0 previ=0 curi=1
handle = 8 , ret_node->handle=65535
handle = 8 , ret_node->handle=16
handle = 8 , ret_node->handle=65535
handle = 8 , ret_node->handle=65535
handle = 8 , ret_node->handle=20
handle = 8 , ret_node->handle=65535
handle = 8 , ret_node->handle=65535
get node by handle fail
subscribe event; conn_handle=0 attr_handle=20 reason=3 prevn=0 curn=0 previ=0 curi=1
handle = 20 , ret_node->handle=65535
handle = 20 , ret_node->handle=16
handle = 20 , ret_node->handle=65535
handle = 20 , ret_node->handle=65535
handle = 20 , ret_node->handle=20
mtu update event; conn_handle=0 cid=4 mtu=203
handle = 17 , ret_node->handle=65535
handle = 17 , ret_node->handle=16
ctxt->op = 2
handle = 18 , ret_node->handle=65535
handle = 18 , ret_node->handle=16
handle = 18 , ret_node->handle=65535
handle = 18 , ret_node->handle=65535
handle = 18 , ret_node->handle=20
handle = 18 , ret_node->handle=65535
handle = 18 , ret_node->handle=65535
get_node_by_att_handle fail , handle[18] 
ctxt->op = 2
handle = 22 , ret_node->handle=65535
handle = 22 , ret_node->handle=16
handle = 22 , ret_node->handle=65535
handle = 22 , ret_node->handle=65535
handle = 22 , ret_node->handle=20
handle = 22 , ret_node->handle=65535
handle = 22 , ret_node->handle=65535
get_node_by_att_handle fail , handle[22] 
ctxt->op = 2
handle = 23 , ret_node->handle=65535
handle = 23 , ret_node->handle=16
handle = 23 , ret_node->handle=65535
handle = 23 , ret_node->handle=65535
handle = 23 , ret_node->handle=20
handle = 23 , ret_node->handle=65535
handle = 23 , ret_node->handle=65535
get_node_by_att_handle fail , handle[23] 
ctxt->op = 2
handle = 22 , ret_node->handle=65535
handle = 22 , ret_node->handle=16
handle = 22 , ret_node->handle=65535
handle = 22 , ret_node->handle=65535
handle = 22 , ret_node->handle=20
handle = 22 , ret_node->handle=65535
handle = 22 , ret_node->handle=65535
get_node_by_att_handle fail , handle[22] 
ctxt->op = 2
handle = 23 , ret_node->handle=65535
handle = 23 , ret_node->handle=16
handle = 23 , ret_node->handle=65535
handle = 23 , ret_node->handle=65535
handle = 23 , ret_node->handle=20
handle = 23 , ret_node->handle=65535
handle = 23 , ret_node->handle=65535
get_node_by_att_handle fail , handle[23] 
ctxt->op = 2
```

ustccw, 2025-03-17:
@Luxin812 PTAL

JHedgesBartec, 2025-09-09:
@Luxin812 as per my original report the ESP32-WROOM32 using the Bluedroid stack running a BLE server correctly reports `+BLECONNPARAM` upon a connection being established, and also when you query the parameters with `AT+BLECONNPARAM?`. 

This issue was raised as the ESP32-C6 does not behave the same, yet the connection parameters are clearly available as they are printed in the debug log for that module.
` conn_itvl=48 conn_latency=0 supervision_timeout=960 encrypted=1 authenticated=1 bonded=1`

closed 2025-12-12 by igrr