espressif/esp-at #912: ESP-AT docs document problem

shiinakaze, 2025-03-02:
I found that the command was incorrectly formatted in the documentation:

```
AT+CWJAP=[<ssid>],[<pwd>][,<bssid>][,<pci_en>][,<reconn_interval>][,<listen_interval>][,<scan_mode>][,<jap_timeout>][,<pmf>]
```

Why is the format of the ssid `<ssid>` and not `<"ssid">` ?

MaJerle, 2025-03-02:
It is a string, as such it has the `"ssid"`

shiinakaze, 2025-03-02:
> It is a string, as such it has the `"ssid"`

I know, but other formats are correct, such as `AT+MQTTPUB=<LinkID>,<"topic">,<"data">,<qos>,<retain>`

ustccw, 2025-03-17:
@shiinakaze We apologize for the confusion caused. All input parameters that are strings should be enclosed in quotes. Our documentation was not clear enough, which led to unnecessary confusion. We will work on improving this section throughout the documentation. Thank you for your feedback!

shiinakaze, 2025-03-17:
> [@shiinakaze](https://github.com/shiinakaze) We apologize for the confusion caused. All input parameters that are strings should be enclosed in quotes. Our documentation was not clear enough, which [...]

谢谢，希望尽快更正。

closed 2025-03-17 by shiinakaze