espressif/esp-at #917: AT+BLEGATTSSRV：GATTS 发现服务指令响应字段说明似乎排序错了

wjsffzb, 2025-03-12:
我的AT版本：
AT version:3.3.1.0-dev(9275716 - ESP32C2 - Sep 19 2024 07:27:28)
SDK version:v5.0.6-dirty
compile time(26a7891):Mar 10 2025 05:26:04
Bin version:v3.3.1.0-dev(ESP32C2-BLE-2MB)
执行"AT+BLEGATTSSRV?"后响应的字段排序似乎是：<start>,<srv_index>,<srv_uuid>,<srv_type>，与官方文档说明不太相符<start>和<srv_index>顺序对调了；

balni23, 2025-03-13:
Same problem with following version
AT version:3.5.0.0-dev(s-33af283 - ESP32C6 - Mar 13 2025 10:44:04)
SDK version:v5.4-dirty
compile time(5dd17454):Mar 13 2025 15:12:05
Bin version:v4.1.0.0-dev(ESP32C6-4MB)

Command answers with error

ustccw, 2025-03-17:
@Luxin812 PTAL

XionHsieh1025, 2025-09-03:
@wjsffzb 

This issue has been resolved in the latest [master](9fad70b9b863d16505a1806e76fc2625361fdeb3) branch.