espressif/esp-at #922: BLE pairing is reoccurring for already paired devices

JHedgesBartec, 2025-03-26:
### Answers checklist

- [x] I have read the documentation [ESP-AT Programming Guide](https://docs.espressif.com/projects/esp-at/en/latest/) and the issue is not addressed there.
- [x] I have used the latest released firmware or have updated my ESP-AT branch (master or release) to the latest version and checked that the issue is present there.
- [x] I have searched the issue tracker for a similar issue and not found a similar issue.

### AT+GMR

AT version:4.0.0.0(3fe3806 - ESP32C6 - Dec 29 2023 11:10:21)
SDK version:v5.1.2-dirty
compile time(89040be7):Jan  2 2024 05:53:07
Bin version:v4.0.0.0(ESP32C6-4MB)

### ESP-AT Firmware Source

Downloaded from: https://docs.espressif.com/projects/esp-at/en/latest/esp32c6/AT_Binary_Lists/esp_at_binaries.html

### Hardware Information

ESP32-C6 WROOM

### Power Supply used

External 3.3V

### What is the expected behavior?

When a BLE client (Any) is pairing to the BLE Server (ESP32), if the client MAC address changes, the previous BLE pairing should not be removed and the connection shall be accepted.

### What is the actual behavior?

When the same BLE Client, in this case an Android tablet, connects to the ESP32 BLE server with a different MAC addresses due to RPA, the existing pairing is being removed and the user has to confirm the pairing to the module again.

### Probability of recurrence

100% when the Android client device changes the MAC address.

### AT+SYSRAM?

Not available but not considered to be memory related.

### Steps to reproduce

1. Create a pairing/bond from a BLE client to the BLE ESP32 Server
2. Wait for the Android device to change its public MAC address, this could be minutes or hours.
3. Connect to the ESP32 server
4. Prompt for pairing is shown

### AT command port output

```plain
ESP-AT << (00086042s) +BLECONN:0,"59:a9:fd:4f:55:65"
ESP-AT << (00086402s) +BLEAUTHCMPL:0,0
ESP-AT << (00111397s) +BLEDISCONN:0,"59:a9:fd:4f:55:65"
ESP-AT >> (00111407s) AT+BLEADVSTART
ESP-AT << (00111416s) 
ESP-AT << (00111425s) OK
ESP-AT << (00114932s) +BLECONN:0,"49:46:6e:2f:cd:b1"
ESP-AT << (00115592s) +BLESECNTFYKEY:0,123456
ESP-AT << (00129899s) +BLEAUTHCMPL:0,0
```

### AT log port output

```plain
ESP-DEBUG >> (00086052s) connection established; status=0 handle=0 our_ota_addr_type=0 our_ota_addr=4083af5f:00:15:4082c000:00:42176c70
ESP-DEBUG >> (00086061s)  our_id_addr_type=0 our_id_addr=4083af51:00:15:4082c000:00:42176c70
ESP-DEBUG >> (00086069s)  peer_ota_addr_type=1 peer_ota_addr=4083af66:00:15:4082c000:00:42176c70
ESP-DEBUG >> (00086081s)  peer_id_addr_type=1 peer_id_addr=4083af58:00:15:4082c000:00:42176c70
ESP-DEBUG >> (00086089s)  conn_itvl=39 conn_latency=0 supervision_timeout=500 encrypted=0 authenticated=0 bonded=0
ESP-DEBUG >> (00086423s) advertise complete; reason=0encryption change event; status=0 handle=0 our_ota_addr_type=0 our_ota_addr=4083aeef:00:15:4082c000:00:42176c70
ESP-DEBUG >> (00086432s)  our_id_addr_type=0 our_id_addr=4083aee1:00:15:4082c000:00:42176c70
ESP-DEBUG >> (00086438s)  peer_ota_addr_type=1 peer_ota_addr=4083aef6:00:15:4082c000:00:42176c70
ESP-DEBUG >> (00086448s)  peer_id_addr_type=1 peer_id_addr=4083aee8:00:15:4082c000:00:42176c70
ESP-DEBUG >> (00086455s)  conn_itvl=39 conn_latency=0 supervision_timeout=500 encrypted=1 authenticated=1 bonded=1
ESP-DEBUG >> (00086465s) 
ESP-DEBUG >> (00086471s) subscribe event; conn_handle=0 attr_handle=8 reason=3 prevn=0 curn=0 previ=0 curi=1
ESP-DEBUG >> (00086482s) handle = 8 , ret_node->handle=65535
ESP-DEBUG >> (00086488s) handle = 8 , ret_node->handle=16
ESP-DEBUG >> (00086499s) handle = 8 , ret_node->handle=65535
ESP-DEBUG >> (00086506s) handle = 8 , ret_node->handle=65535
ESP-DEBUG >> (00086517s) handle = 8 , ret_node->handle=20
ESP-DEBUG >> (00086523s) handle = 8 , ret_node->handle=65535
ESP-DEBUG >> (00086534s) handle = 8 , ret_node->handle=65535
ESP-DEBUG >> (00086540s) get node by handle fail
ESP-DEBUG >> (00086674s) handle = 17 , ret_node->handle=65535
ESP-DEBUG >> (00086681s) handle = 17 , ret_node->handle=16
ESP-DEBUG >> (00086688s) ctxt->op = 2
ESP-DEBUG >> (00086779s) handle = 22 , ret_node->handle=65535
ESP-DEBUG >> (00086786s) handle = 22 , ret_node->handle=16
ESP-DEBUG >> (00086793s) handle = 22 , ret_node->handle=65535
ESP-DEBUG >> (00086804s) handle = 22 , ret_node->handle=65535
ESP-DEBUG >> (00086811s) handle = 22 , ret_node->handle=20
ESP-DEBUG >> (00086823s) handle = 22 , ret_node->handle=65535
ESP-DEBUG >> (00086830s) handle = 22 , ret_node->handle=65535
ESP-DEBUG >> (00086840s) get_node_by_att_handle fail , handle[22] 
ESP-DEBUG >> (00086848s) ctxt->op = 2
ESP-DEBUG >> (00111269s) subscribe event; conn_handle=0 attr_handle=8 reason=2 prevn=0 curn=0 previ=1 curi=0
ESP-DEBUG >> (00111276s) handle = 8 , ret_node->handle=65535
ESP-DEBUG >> (00111282s) handle = 8 , ret_node->handle=16
ESP-DEBUG >> (00111292s) handle = 8 , ret_node->handle=65535
ESP-DEBUG >> (00111298s) handle = 8 , ret_node->handle=65535
ESP-DEBUG >> (00111308s) handle = 8 , ret_node->handle=20
ESP-DEBUG >> (00111314s) handle = 8 , ret_node->handle=65535
ESP-DEBUG >> (00111325s) handle = 8 , ret_node->handle=65535
ESP-DEBUG >> (00111331s) get node by handle fail
ESP-DEBUG >> (00111341s) disconnect; reason=531 handle=0 our_ota_addr_type=0 our_ota_addr=4083afbb:00:15:4082c000:00:42176c70
ESP-DEBUG >> (00111349s)  our_id_addr_type=0 our_id_addr=4083afad:00:15:4082c000:00:42176c70
ESP-DEBUG >> (00111362s)  peer_ota_addr_type=1 peer_ota_addr=4083afc2:00:15:4082c000:00:42176c70
ESP-DEBUG >> (00111369s)  peer_id_addr_type=1 peer_id_addr=4083afb4:00:15:4082c000:00:42176c70
ESP-DEBUG >> (00111381s)  conn_itvl=39 conn_latency=0 supervision_timeout=500 encrypted=1 authenticated=1 bonded=1
ESP-DEBUG >> (00111387s) 
ESP-DEBUG >> (00114916s) connection established; status=0 handle=0 our_ota_addr_type=0 our_ota_addr=4083af5f:00:15:4082c000:00:42176c70
ESP-DEBUG >> (00114942s)  our_id_addr_type=0 our_id_addr=4083af51:00:15:4082c000:00:42176c70
ESP-DEBUG >> (00114952s)  peer_ota_addr_type=1 peer_ota_addr=4083af66:00:15:4082c000:00:42176c70
ESP-DEBUG >> (00114961s)  peer_id_addr_type=1 peer_id_addr=4083af58:00:15:4082c000:00:42176c70
ESP-DEBUG >> (00114974s)  conn_itvl=39 conn_latency=0 supervision_timeout=500 encrypted=0 authenticated=0 bonded=0
ESP-DEBUG >> (00115599s) advertise complete; reason=0PASSKEY_ACTION_EVENT started 
ESP-DEBUG >> (00129886s) encryption change event; status=0 handle=0 our_ota_addr_type=0 our_ota_addr=4083aeef:00:15:4082c000:00:42176c70
ESP-DEBUG >> (00129909s)  our_id_addr_type=0 our_id_addr=4083aee1:00:15:4082c000:00:42176c70
ESP-DEBUG >> (00129918s)  peer_ota_addr_type=1 peer_ota_addr=4083aef6:00:15:4082c000:00:42176c70
ESP-DEBUG >> (00129926s)  peer_id_addr_type=1 peer_id_addr=4083aee8:00:15:4082c000:00:42176c70
ESP-DEBUG >> (00129940s)  conn_itvl=39 conn_latency=0 supervision_timeout=500 encrypted=1 authenticated=1 bonded=1
ESP-DEBUG >> (00129947s) 
ESP-DEBUG >> (00130549s) handle = 17 , ret_node->handle=65535
ESP-DEBUG >> (00130556s) handle = 17 , ret_node->handle=16
ESP-DEBUG >> (00130562s) ctxt->op = 2
ESP-DEBUG >> (00130751s) handle = 22 , ret_node->handle=65535
ESP-DEBUG >> (00130757s) handle = 22 , ret_node->handle=16
ESP-DEBUG >> (00130763s) handle = 22 , ret_node->handle=65535
ESP-DEBUG >> (00130775s) handle = 22 , ret_node->handle=65535
ESP-DEBUG >> (00130782s) handle = 22 , ret_node->handle=20
ESP-DEBUG >> (00130794s) handle = 22 , ret_node->handle=65535
ESP-DEBUG >> (00130800s) handle = 22 , ret_node->handle=65535
ESP-DEBUG >> (00130812s) get_node_by_att_handle fail , handle[22] 
ESP-DEBUG >> (00130818s) ctxt->op = 2
ESP-DEBUG >> (00131353s) subscribe event; conn_handle=0 attr_handle=8 reason=1 prevn=0 curn=0 previ=0 curi=1
ESP-DEBUG >> (00131360s) handle = 8 , ret_node->handle=65535
ESP-DEBUG >> (00131366s) handle = 8 , ret_node->handle=16
ESP-DEBUG >> (00131377s) handle = 8 , ret_node->handle=65535
ESP-DEBUG >> (00131383s) handle = 8 , ret_node->handle=65535
ESP-DEBUG >> (00131396s) handle = 8 , ret_node->handle=20
ESP-DEBUG >> (00131402s) handle = 8 , ret_node->handle=65535
ESP-DEBUG >> (00131413s) handle = 8 , ret_node->handle=65535
ESP-DEBUG >> (00131419s) get node by handle fail
```

### More Information.

"59:a9:fd:4f:55:65" and "49:46:6e:2f:cd:b1" are the same physical Android tablet running the same application.

Luxin812, 2025-09-09:
Please test your scenario using the latest released ESP32C6 AT [Release v4.1.1.0](https://dl.espressif.com/esp-at/firmwares/esp32c6/ESP32-C6-4MB-AT-V4.1.1.0.zip) to confirm whether the issue still exists. Thank you!

Luxin812, 2025-12-10:
@JHedgesBartec I don't quite understand why the two issues you mentioned above would affect your testing to see if this problem has been resolved. Could you please modify your testing process and first use ESP32C6 AT [Release v4.1.1.0](https://dl.espressif.com/esp-at/firmwares/esp32c6/ESP32-C6-4MB-AT-V4.1.1.0.zip) to confirm if the problem still exists? You can follow up on the other two issues separately on GitHub. Regarding your mention of receiving a fix from sales@espressif.com to resolve issue 2, could you please track the merge status of the fix provided by sales@espressif.com at that time, or update the version information (AT+GMR) of the fix you received in #979 so that we can understand the progress? thanks.

XionHsieh1025, 2025-12-24:
@JHedgesBartec 

The attached library is a temporary build. I have verified locally that it works as expected.
If it also works correctly in your scenario, we will merge this fix immediately.

[99101577.zip](https://github.com/user-attachments/files/24328742/99101577.zip)

JHedgesBartec, 2026-01-06:
@XionHsieh1025 

Which commit of ESP-AT and ESP-IDF should I use to build this with?

JHedgesBartec, 2026-01-16:
@XionHsieh1025 any suggestions?

JHedgesBartec, 2026-02-13:
@ustccw any progress on this internally?