espressif/esp-at #933: AT+CWLAP returns empty on ESP32-C6

nmaggioni, 2025-04-15:
### Answers checklist

- [x] I have read the documentation [ESP-AT Programming Guide](https://docs.espressif.com/projects/esp-at/en/latest/) and the issue is not addressed there.
- [x] I have used the latest released firmware or have updated my ESP-AT branch (master or release) to the latest version and checked that the issue is present there.
- [x] I have searched the issue tracker for a similar issue and not found a similar issue.

### AT+GMR

```
AT+GMR
AT version:4.0.0.0(3fe3806 - ESP32C6 - Dec 29 2023 11:10:21)
SDK version:v5.1.2-dirty
compile time(89040be7):Jan  2 2024 05:53:07                                                             
Bin version:v4.0.0.0(ESP32C6-4MB)                                                                       
                                                                                                        
OK                                                                                                      
```

### ESP-AT Firmware Source

Downloaded from the [docs page](https://docs.espressif.com/projects/esp-at/en/latest/esp32c6/AT_Binary_Lists/esp_at_binaries.html) ([direct link](https://dl.espressif.com/esp-at/firmwares/esp32c6/ESP32-C6-4MB-AT-V4.0.0.0.zip)).

### Hardware Information

ESP32-C6-WROOM-1 module on custom board.

### Power Supply used

External 3.3V

### What is the expected behavior?

Sending `AT+CWLAP` should return a list of visible APs, followed by `OK`.

### What is the actual behavior?

Sending `AT+CWLAP` only returns `OK`.

### Probability of recurrence

Happens 100% of the time.

### AT+SYSRAM?

```
AT+SYSRAM?                                                                                              
+SYSRAM:259088,50244                                                                                    
OK
```

### Steps to reproduce

1. Boot up the module.
2. Wait for it to report `ready` through the AT UART.
3. Send the `AT+CWMODE=1` if it's the first boot after flashing.
4. Send the `AT+CWLAP` command.

### AT command port output

```plain
AT+CWMODE=1

OK
AT+CWLAP

OK
```

### AT log port output

```plain
ESP-ROM:esp32c6-20220919
Build:Sep 19 2022
rst:0xc (SW_CPU),boot:0x4f (SPI_FAST_FLASH_BOOT)
Saved PC:0x4001975a
SPIWP:0xee
mode:DIO, clock div:2
load:0x4086c410,len:0xd2c
load:0x4086e610,len:0x2df0
load:0x40875728,len:0x17d8
entry 0x4086c410
(22) boot: ESP-IDF v5.1.2-dirty 2nd stage bootloader
(23) boot: compile time Jan  2 2024 05:53:08
(24) boot: chip revision: v0.0
(25) boot.esp32c6: SPI Speed      : 80MHz
(30) boot.esp32c6: SPI Mode       : DIO
(35) boot.esp32c6: SPI Flash Size : 4MB
(40) boot: Enabling RNG early entropy source...
(45) boot: Partition Table:
(48) boot: ## Label            Usage          Type ST Offset   Length
(56) boot:  0 otadata          OTA data         01 00 0000d000 00002000
(63) boot:  1 phy_init         RF data          01 01 0000f000 00001000
(71) boot:  2 nvs              WiFi data        01 02 00010000 0000e000
(78) boot:  3 at_customize     unknown          40 00 0001e000 00042000
(86) boot:  4 ota_0            OTA app          00 10 00060000 001d0000
(93) boot:  5 ota_1            OTA app          00 11 00230000 001d0000
(101) boot: End of partition table
(105) esp_image: segment 0: paddr=00060020 vaddr=42150020 size=30740h (198464) map
(195) esp_image: segment 1: paddr=00090768 vaddr=40800000 size=0f8b0h ( 63664) load
(225) esp_image: segment 2: paddr=000a0020 vaddr=42000020 size=146000h (1335296) map
(770) esp_image: segment 3: paddr=001e6028 vaddr=4080f8b0 size=0a17ch ( 41340) load
(790) esp_image: segment 4: paddr=001f01ac vaddr=40819a30 size=03bf8h ( 15352) load
(799) esp_image: segment 5: paddr=001f3dac vaddr=50000000 size=00068h (   104) load
(812) boot: Loaded app from partition at offset 0x60000
(813) boot: Disabling RNG early entropy source...
no external 32k oscillator, disable it now.

at param mode: 1

error: pll_cal exceeds 2ms!!!
error: pll_cal exceeds 2ms!!!
AT cmd port:uart1 tx:7 rx:6 cts:5 rts:4 baudrate:115200

module_name: ESP32C6-4MB

max tx power=78, ret=0

v4.0.0.0
```

### More Information.

Running the [WiFiScan Arduino example](https://github.com/espressif/arduino-esp32/blob/f1223663dd122b605603c1e072a33f3a6935f451/libraries/WiFi/examples/WiFiScan/WiFiScan.ino) reliably shows the available networks as expected, so this does not seem to be a hardware issue.

Flashing the [most recent](https://github.com/espressif/esp-at/actions/runs/14210401779) esp-at build from the `master` branch results in random reboots.

nmaggioni, 2025-04-16:
Manually bumping ESP-IDF to commit [e9bdd39](https://github.com/espressif/esp-idf/commit/e9bdd395995f60878e01cece10525987bacc8a5f) seems to solve the issue for me.

ustccw, 2025-04-18:
This might be a [power issue](https://github.com/espressif/arduino-esp32/issues/9411#issuecomment-2016864332).

Could you please check your power supply.

nmaggioni, 2025-04-18:
The reboots are indeed most probably caused by a suboptimal regulator choice, thanks. As a side note, changing the brownout detector threshold does not seem to affect this issue.

But the AP list coming out empty is a pretty consistent behavior nonetheless: by either updating ESP-IDF or running the Arduino example sketch I get the list correctly populated every time on the very same hardware.

ustccw, 2025-04-21:
To verify whether this is a software or hardware issue, you can run this example to see if it works correctly:
https://github.com/espressif/esp-idf/tree/v5.4.1/examples/wifi/scan

JHedgesBartec, 2025-05-06:
@nmaggioni have you tried modifying the mfg_nvs partition to set the max RF power there?

[Generate mfg_nvs.bin](https://docs.espressif.com/projects/esp-at/en/latest/esp32/Compile_and_Develop/How_to_update_mfg_nvs.html#generate-mfg-nvs-bin)

nmaggioni, 2025-05-06:
@JHedgesBartec I did, but with the released AT firmware the AP list comes up empty nonetheless.

ustccw, 2025-05-07:
This might be a misunderstanding.
```
ESP_ERROR_CHECK(esp_wifi_set_max_tx_power(40));
```

To achieve better TX power, you should set a higher value, such as 80.
https://docs.espressif.com/projects/esp-at/en/latest/esp32c6/AT_Command_Set/Basic_AT_Commands.html#id91

nmaggioni, 2025-05-07:
@ustccw Thanks, but that change was indeed made to lower the current draw and try to exclude any power related issues. If I'm not mistaken, the WiFi scan sequence [does not involve any transmitting anyway](https://docs.espressif.com/projects/esp-idf/en/v5.4.1/esp32c6/api-guides/wifi.html#scan-all-aps-on-all-channels-foreground), so lowering the TX power should not affect the AP scan results.

I should be able to retest the behavior of a new board with a beefier regulator in the coming weeks.

ustccw, 2025-05-08:
Generally speaking, setting a lower transmission power does reduce power consumption and won't affect the scanning process, but it may result in fewer scan results.

nmaggioni, 2025-05-09:
I don't think the power setting is the issue here: as I mentioned before the ESP-IDF and Arduino examples work fine, it's only the ESP-AT firmware that reports no APs found; bumping the ESP-IDF core version solves the problem, see above.

ustccw, 2025-05-23:
@nmaggioni Could you please enable debug log and wpa_supplicant log to see what happened?