espressif/esp-at #935: BT Classic fixed pin is ignored

JHedgesBartec, 2025-04-23:
### Answers checklist

- [x] I have read the documentation [ESP-AT Programming Guide](https://docs.espressif.com/projects/esp-at/en/latest/) and the issue is not addressed there.
- [x] I have used the latest released firmware or have updated my ESP-AT branch (master or release) to the latest version and checked that the issue is present there.
- [x] I have searched the issue tracker for a similar issue and not found a similar issue.

### AT+GMR

AT+GMR
AT version:3.5.0.0-dev(s-ba52610 - ESP32 - Oct 31 2024 08:02:56)
SDK version:v5.0.6-dirty
compile time(a5f8f393):Nov 19 2024 14:45:45
Bin version:v4.0.0.1(WROOM-32)

OK

### ESP-AT Firmware Source

Downloaded from Master, with commit reverted to `102d5127`
BT Classic Enabled.

### Hardware Information

ESP32-WROOM-32

### Power Supply used

USB

### What is the expected behavior?

If a fixed pin is set, if a client attempts to pair and connect, the fixed pin is used.

### What is the actual behavior?

If a client connects, the fixed pin is ignored.

### Probability of recurrence

100%

### AT+SYSRAM?

AT+SYSRAM?
+SYSRAM:82300,75040
OK

### Steps to reproduce

1. Initialise BT Classic
2. Set a Fixed Pin with no IO capability
3. Set the scan mode to discoverable and connectable
4. Use Windows or Android device to initiate BT classic pairing

### AT command port output

```plain
ready
ATE0
ATE0

OK
AT+GMR
AT version:3.5.0.0-dev(s-ba52610 - ESP32 - Oct 31 2024 08:02:56)
SDK version:v5.0.6-dirty
compile time(a5f8f393):Nov 19 2024 14:45:45
Bin version:v4.0.0.1(WROOM-32)

OK
AT+RFPOWER=60

OK
AT+BLEINIT=0

OK
AT+BTINIT=1

OK
AT+BTNAME="ESP32DevKitC"

OK
AT+BTSECPARAM=3,1,"ABCD1234"

OK
AT+BTADDR?
+BTADDR:"90:38:0c:ed:67:02"

OK
AT+BTSPPINIT=1

OK
AT+BTSCANMODE=2

OK
AT+CWINIT=1

OK
AT+CWMODE=1

OK
AT+CIPSTAMAC?
+CIPSTAMAC:"90:38:0c:ed:67:00"

OK
AT+SYSRAM?
+SYSRAM:82300,75040
OK
```

### AT log port output

```plain
Silence mode enabled to allow compilation wit BT classic.
```

### More Information.

I have tried disabling BT Classic SSP through menu config in case this overrides the fixed pin, however the compilation fails with undefined references to functions in at_btCmd.c.obj.

XionHsieh1025, 2026-01-04:
Currently, Classic Bluetooth AT does not support disabling Secure Connections (SC).

JHedgesBartec, 2026-01-05:
For reference of other users, this can be done by disabling SSP through menu config then patching `esp-at/esp-idf/components/bt/host/bluedroid/device/controller.c` to set `controller_param.simple_pairing_supported` to false.

closed 2026-01-05 by JHedgesBartec