espressif/esp-at #940: Crash on boot of ESP32 C6 when updating to latest master

JHedgesBartec, 2025-05-23:
### Answers checklist

- [x] I have read the documentation [ESP-AT Programming Guide](https://docs.espressif.com/projects/esp-at/en/latest/) and the issue is not addressed there.
- [x] I have used the latest released firmware or have updated my ESP-AT branch (master or release) to the latest version and checked that the issue is present there.
- [x] I have searched the issue tracker for a similar issue and not found a similar issue.

### AT+GMR

AT+GMR
AT version:3.5.0.0-dev(2a11c83 - ESP32C6 - May  8 2025 11:25:55)
SDK version:v5.4.1-643-g8ad0d3d8f2-dirty
compile time(0b787ec5):May 23 2025 12:45:11
Bin version:v1.1.0.0(ESP32C6-4MB)

OK

### ESP-AT Firmware Source

Download from master with minor changes to AT such as disabling WPS, and other features not required.

### Hardware Information

ESP32-C6-WROOM-1

### Power Supply used

External 3.3V

### What is the expected behavior?

Module boots

### What is the actual behavior?

Module crashes

### Probability of recurrence

100%

### AT+SYSRAM?

N/A

### Steps to reproduce

Unsure of certain steps before, however the module was programmed using a factory image based on master commit 5302cdc7
The module was then updated using [esp-serial-flasher](https://github.com/espressif/esp-serial-flasher) and a STM32 microcontroller, programming the binaries as explained [here](https://github.com/espressif/esp-serial-flasher)
Once programmed and the module reboots, the module crashes on every single boot thereafter.

The module has been used for BLE only.

When reading out the flash memory, the differences are in the 10000 to 1DFD8 address range when compared with the factory image.

Original factory image:
[esp_c6_at_factory_1_0_2_2.zip](https://github.com/user-attachments/files/20413616/esp_c6_at_factory_1_0_2_2.zip)

This is the newly compiled software, containing the components, new factory image and 'flash_contents_post_update.bin' for comparison:
[master_build.zip](https://github.com/user-attachments/files/20413581/files.zip)

I have tried performing the same update using esptool and a devkit however this does not appear to have the same issue. Though this would not have had the same pairings or configuration as the module.

### AT command port output

```plain
No command port as module is crashing
```

### AT log port output

```plain
ESP-ROM:esp32c6-20220919
Build:Sep 19 2022
rst:0x1 (POWERON),boot:0x6f (SPI_FAST_FLASH_BOOT)
SPIWP:0xee
mode:DIO, clock div:2
load:0x40875720,len:0x16b8
load:0x4086c110,len:0xe70
load:0x4086e610,len:0x30e8
entry 0x4086c11a
I (23) boot: ESP-IDF v5.4.1-643-g8ad0d3d8f2-dirty 2nd stage bootloader
I (24) boot: compile time May 23 2025 12:44:36
I (24) boot: chip revision: v0.0
I (25) boot: efuse block revision: v0.1
I (29) boot.esp32c6: SPI Speed      : 80MHz
I (33) boot.esp32c6: SPI Mode       : DIO
I (37) boot.esp32c6: SPI Flash Size : 4MB
I (40) boot: Enabling RNG early entropy source...
I (45) boot: Partition Table:
I (48) boot: ## Label            Usage          Type ST Offset   Length
I (54) boot:  0 otadata          OTA data         01 00 0000d000 00002000
I (60) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (67) boot:  2 nvs              WiFi data        01 02 00010000 0000e000
I (73) boot:  3 at_customize     unknown          40 00 0001e000 00042000
I (80) boot:  4 ota_0            OTA app          00 10 00060000 001d0000
I (86) boot:  5 ota_1            OTA app          00 11 00230000 001d0000
I (93) boot: End of partition table
I (96) esp_image: segment 0: paddr=00060020 vaddr=42160020 size=2a774h (173940) map
I (136) esp_image: segment 1: paddr=0008a79c vaddr=40800000 size=0587ch ( 22652) load
I (141) esp_image: segment 2: paddr=00090020 vaddr=42000020 size=1599e4h (1415652) map
I (403) esp_image: segment 3: paddr=001e9a0c vaddr=4080587c size=1afd8h (110552) load
I (427) esp_image: segment 4: paddr=002049ec vaddr=40820860 size=04530h ( 17712) load
I (431) esp_image: segment 5: paddr=00208f24 vaddr=50000000 size=0005ch (    92) load
I (432) esp_image: segment 6: paddr=00208f88 vaddr=5000005c size=00020h (    32) load
I (446) boot: Loaded app from partition at offset 0x60000
I (446) boot: Disabling RNG early entropy source...
I (677) at-init: at param mode: 1
I (739) at-uart: AT cmd port:uart1 tx:7 rx:6 cts:5 rts:4 baudrate:115200
I (740) at-init: module_name: ESP32C6-4MB
I (741) at-init: max tx power=78, ret=0
I (743) at-init: v1.1.0.0 (unknown)

assert failed: block_locate_free heap_tlsf.c:476 (block_size(block) >= size)
Core  0 register dump:
MEPC    : 0x408009b8  RA      : 0x40810e82  SP      : 0x40832ed0  GP      : 0x40821064  
TP      : 0x40833160  T0      : 0x37363534  T1      : 0x7271706f  T2      : 0x33323130  
S0/FP   : 0x0000007b  S1      : 0x00000001  A0      : 0x40832f0c  A1      : 0x40824085  
A2      : 0x00000001  A3      : 0x00000029  A4      : 0x00000001  A5      : 0x4082e000  
A6      : 0x0000000c  A7      : 0x76757473  S2      : 0x00000009  S3      : 0x4083302c  
S4      : 0x40824084  S5      : 0x00000000  S6      : 0x00000000  S7      : 0x00000000  
S8      : 0x00000000  S9      : 0x00000000  S10     : 0x00000000  S11     : 0x00000000  
T3      : 0x6e6d6c6b  T4      : 0x6a696867  T5      : 0x66656463  T6      : 0x62613938  
MSTATUS : 0x00001881  MTVEC   : 0x40800001  MCAUSE  : 0x00000002  MTVAL   : 0x00000000  
MHARTID : 0x00000000  

Stack memory:
40832ed0: 0x0000000c 0x4004b8d8 0x4004d078 0x40816038 0x40824068 0x4004d078 0x40824078 0x4004d150
40832ef0: 0x4082407c 0x40832f04 0x40824080 0x4004d344 0x40824084 0x00363734 0x40832fbc 0x65737361
40832f10: 0x66207472 0x656c6961 0x62203a64 0x6b636f6c 0x636f6c5f 0x5f657461 0x65657266 0x61656820
40832f30: 0x6c745f70 0x632e6673 0x3637343a 0x6c622820 0x5f6b636f 0x657a6973 0x6f6c6228 0x20296b63
40832f50: 0x73203d3e 0x29657a69 0x69726500 0x6369646f 0x7461645f 0x00000061 0xffffffff 0xffffffff
40832f70: 0x006e655f 0x00000000 0xffffffff 0x00000000 0x40811128 0x408111ba 0x40833000 0x40821064
40832f90: 0x40833160 0x40022494 0xffffffe0 0xffffffff 0x40830550 0x4084ca5c 0x00000001 0x4084ca5c
40832fb0: 0x4084ca5c 0x00000001 0x00000000 0x20001000 0x00000003 0x000000ff 0x40880000 0x00000000
40832fd0: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x0000001a 0x00000354
40832ff0: 0x0000000b 0x40830564 0x4084d1e4 0x40028392 0x0000000b 0x40830564 0x4084d1e4 0x400213e2
40833010: 0x00000000 0x00000000 0x00000000 0x00000000 0x00001800 0x40880000 0x40830550 0x40022542
40833030: 0x00000000 0x00000001 0x4082dee8 0x00000354 0x00001800 0x00000000 0x40830c30 0x40800cc0
40833050: 0x00001800 0x00000354 0x42160168 0x40800dac 0x08080808 0x00001800 0x00000354 0x00000004
40833070: 0x00000000 0x00000354 0x42160168 0x40800de2 0x40880000 0x4084ce94 0x40830550 0x40800a8e
40833090: 0x00000000 0x00000354 0x42160168 0x40800aea 0x00000000 0x00000000 0x42160168 0x40816042
408330b0: 0x00000000 0x00000000 0x4084ce94 0x42014c4e 0x00000000 0x00000000 0x42160168 0x42015bb4
408330d0: 0x42163f0c 0x4217cb08 0x42160140 0x42011b84 0x07d2f0e2 0x00000000 0x00000000 0x42011aea
408330f0: 0x00000000 0x4083437c 0x40833114 0x4201145a 0x00000000 0x00000000 0x00000000 0x42155f00
40833110: 0x00000000 0x0000ea60 0x00000001 0x00000001 0x00000000 0x00000000 0x00000000 0x408110ae
40833130: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
40833150: 0x00000000 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0x00000150 0x40832f80 0x00000039
40833170: 0x40824fc8 0x40824fc8 0x40833168 0x40824fc0 0x00000018 0x00000000 0x00000000 0x40833168
40833190: 0x00000000 0x00000001 0x40832364 0x6e69616d 0x00000000 0x00000000 0x00000000 0x40833160
408331b0: 0x00000001 0x00000000 0x40839288 0x4200a26e 0x00000000 0x40830ea0 0x40830f08 0x40830f70
408331d0: 0x00000000 0x00000000 0x00000001 0x00000000 0x00000000 0x00000000 0x4002849c 0x00000000
408331f0: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
40833210: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
40833230: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
40833250: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
40833270: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
40833290: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
408332b0: 0x00000000 0x00000000 0x00000600 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5

ELF file SHA256: e7002aa42

Rebooting...
```

JHedgesBartec, 2025-05-27:
@ustccw our gatts data has not changed since the previous release.

We were previously building from commit 5302cdc7 with a patched libesp32c6_at_core.a from your support team such that the characteristics could be setup to require pairings. This is because there was a bug with handling the permissions for characteristics with the Nimble stack.

Here is our gatts data:
```
# index,uuid_len,uuid,perm,max_len,cur_len,value
0,16,0x2800,0x01,2,2,BA1D
1,16,0x2803,0x01,1,1,1008
2,16,0x1000,0x10,512,512,
3,16,0x2901,0x01,8,8,5472616E736D6974
4,16,0x2904,0x01,7,7,19000000010000
5,16,0x2803,0x01,1,1,20
6,16,0x1001,0x01,1,1,00
7,16,0x2901,0x01,7,7,52656365697665
8,16,0x2902,0x11,2,2,0000
9,16,0x2904,0x01,7,7,19000000010000
```

I have flashed the ESP32-C6-DevkitC with the same binary contents of the ESP32-C6-WROOM module I was testing with, then updated using the components (not factory image) using the flash download tool and can confirm the same bootloop issue. Again this points to some previous state on the initial software build.

As requested I have performed the same test, but this time updating to master using the provided gatts data, the device still crashes and boot-loops with the same assert of `assert failed: block_locate_free heap_tlsf.c:476 (block_size(block) >= size)`

All I have changed from master commit 0b787ec5 is the [sdkconfig.txt](https://github.com/user-attachments/files/20454396/sdkconfig.txt) and [gatts_data.csv](https://github.com/user-attachments/files/20454394/gatts_data.csv).

JHedgesBartec, 2025-05-27:
Here is a image of the full 4MB flash before the update, flash the module with this image before you update the components.
[esp32_c6_wroom_module_image.zip](https://github.com/user-attachments/files/20454770/esp32_c6_wroom_module_image.zip)

You can then use the flash download tool to update the components as follows:
[image]

Performing these steps show the crash.

ustccw, 2025-05-27:
@JHedgesBartec Thanks for your feedback. I can reproduce this issue now. Will look into it this week.

ustccw, 2025-05-28:
@JHedgesBartec Thank you for the test. The fix should be auto sync to GitHub this week.

closed 2025-05-29 by igrr

ustccw, 2025-05-29:
@JHedgesBartec Hi! This issue should be fixed on latest master. Please check. If not, just reopen the issue again.