espressif/esp-at #953: Crash on ESP32 when sending BLE data from Server to Client

JHedgesBartec, 2025-07-25:
### Answers checklist

- [x] I have read the documentation [ESP-AT Programming Guide](https://docs.espressif.com/projects/esp-at/en/latest/) and the issue is not addressed there.
- [x] I have used the latest released firmware or have updated my ESP-AT branch (master or release) to the latest version and checked that the issue is present there.
- [x] I have searched the issue tracker for a similar issue and not found a similar issue.

### AT+GMR

AT+GMR
AT+GMR
AT version:4.2.0.0-dev(s-92a6ba3 - ESP32 - Jul 23 2025 02:46:54)
SDK version:v5.4.1-643-g8ad0d3d8f2-dirty
compile time(dc213132):Jul 24 2025 12:17:13
Bin version:v5.0.0.0(WROOM-32)

OK

### ESP-AT Firmware Source

Master cloned from commit [dc21313](https://github.com/espressif/esp-at/commit/dc21313202361c23e1b418480317e13140ebe472)
With modifications to gatts_data some AT features removed (non BLE) and BT Classic enabled.

### Hardware Information

ESP32-WROOM-32

### Power Supply used

External 3.3V

### What is the expected behavior?

Data is sent from BLE ESP32 server to client without issue.

### What is the actual behavior?

When sending data from the BLE ESP32 server to the client, there is no response to the `AT+BLEGATTSIND` command and the module eventually crashes.
Comparing against our last build of master (102d5127) the only difference from the command and port trace is the lack of printing BLE service information after the `AT+BLEGATTSSRVCRE` command.
The module is initialized in an identical way.

### Probability of recurrence

100%

### AT+SYSRAM?

Not available

### Steps to reproduce

Configure a BLE ESP32 Server using a fixed pin.
Connect a client and pair.
Send data from server to client
No response followed by a crash/reboot.

### AT command port output

```plain
ESP-AT << (00005124s) 
ESP-AT << (00005136s) ready
ESP-AT >> (00005150s) ATE0
ESP-AT << (00005163s) ATE0
ESP-AT << (00005177s) 
ESP-AT << (00005189s) OK
ESP-AT >> (00005204s) AT+GMR
ESP-AT << (00005221s) AT version:4.2.0.0-dev(s-92a6ba3 - ESP32 - Jul 23 2025 02:46:54)
ESP-AT << (00005238s) SDK version:v5.4.1-643-g8ad0d3d8f2-dirty
ESP-AT << (00005256s) compile time(dc213132):Jul 24 2025 12:17:13
ESP-AT << (00005276s) Bin version:v5.0.0.0(WROOM-32)
ESP-AT << (00005293s) 
[... 157 lines cut ...]
ESP-DEBUG >> (00153260s) I (48) boot: Enabling RNG early entropy source...
ESP-DEBUG >> (00153281s) I (52) boot: Partition Table:
ESP-DEBUG >> (00153298s) I (55) boot: ## Label            Usage          Type ST Offset   Length
ESP-DEBUG >> (00153322s) I (61) boot:  0 phy_init         RF data          01 01 0000f000 00001000
ESP-DEBUG >> (00153345s) I (67) boot:  1 otadata          OTA data         01 00 00010000 00002000
ESP-DEBUG >> (00153368s) I (74) boot:  2 nvs              WiFi data        01 02 00012000 0000e000
ESP-DEBUG >> (00153391s) I (80) boot:  3 at_customize     unknown          40 00 00020000 000e0000
ESP-DEBUG >> (00153420s) I (87) boot:  4 ota_0            OTA app          00 10 00100000 00180000
ESP-DEBUG >> (00153445s) I (93) boot:  5 ota_1            OTA app          00 11 00280000 00180000
ESP-DEBUG >> (00153469s) I (100) boot: End of partition table
ESP-DEBUG >> (00153491s) I (103) esp_image: segment 0: paddr=00100020 vaddr=3f400020 size=132e8h ( 78568) map
ESP-DEBUG >> (00153518s) I (138) esp_image: segment 1: paddr=00113310 vaddr=3ff80064 size=0001ch (    28) load
ESP-DEBUG >> (00153548s) I (138) esp_image: segment 2: paddr=00113334 vaddr=3ffbdb60 size=05610h ( 22032) load
ESP-DEBUG >> (00153574s) I (150) esp_image: segment 3: paddr=0011894c vaddr=40080000 size=076cch ( 30412) load
ESP-DEBUG >> (00153605s) I (162) esp_image: segment 4: paddr=00120020 vaddr=400d0020 size=134cf0h (1264880) map
ESP-DEBUG >> (00153630s) I (593) esp_image: segment 5: paddr=00254d18 vaddr=400876cc size=14040h ( 81984) load
ESP-DEBUG >> (00153657s) I (625) esp_image: segment 6: paddr=00268d60 vaddr=400c0000 size=00064h (   100) load
ESP-DEBUG >> (00153689s) I (639) boot: Loaded app from partition at offset 0x100000
ESP-DEBUG >> (00153712s) I (639) boot: Disabling RNG early entropy source...
ESP-DEBUG >> (00153731s) I (876) at-init: at param mode: 1
ESP-DEBUG >> (00153749s) I (967) at-uart: AT cmd port:uart1 tx:17 rx:16 cts:15 rts:14 baudrate:921600
ESP-DEBUG >> (00153771s) I (968) at-init: module_name: WROOM-32
ESP-DEBUG >> (00153792s) I (970) at-init: max tx power=60, ret=0
ESP-DEBUG >> (00153810s) I (972) at-init: v5.0.0.0 (unknown)
```

JHedgesBartec, 2025-08-28:
@ritesh006 I have not tested this since master commit [dc21313](https://github.com/espressif/esp-at/commit/dc21313202361c23e1b418480317e13140ebe472) though there is nothing in the commit notes since to suggest it has been fixed.

Are you seeing the same issue?

XionHsieh1025, 2025-09-03:
Hi @JHedgesBartec  

I tried to reproduce this issue, but failed. 

The firmware for the test was manually compiled by me. By default, if the BT command is enabled, the firmware will exceed the size of the partition table. Therefore, I turned off some functions.

```c
[09:58:13:947] AT+GMR␍␊
[09:58:13:947] AT version:4.2.0.0-dev(s-92a6ba3 - ESP32 - Jul 23 2025 02:46:54)␍␊
[09:58:13:947] SDK version:v5.4.1-643-g8ad0d3d8f2f-dirty␍␊
[09:58:13:947] compile time(dc213132):Sep  3 2025 09:54:56␍␊
[09:58:13:947] Bin version:v4.2.0.0-dev(WROOM-32)␍␊
[09:58:13:947] ␍␊
[09:58:13:947] OK␍␊
```

From the log you posted, I found that the char index you sent for indication is 2. Therefore, I guess you modified the service table. If that's the case, I hope you can share your modifications.

The attachment is my test firmware and test log. I hope you can check if there are any differences between my operation and yours.

[esp-at-issue-953(2025.9.3).zip](https://github.com/user-attachments/files/22108024/esp-at-issue-953.2025.9.3.zip)