espressif/esp-at #962: Wemos D1 mini (8266EX). Unable to receive any AT response. Not sure what firmware to flash either

5iemen5, 2025-09-03:
### Answers checklist

- [x] I have read the documentation [ESP-AT Programming Guide](https://docs.espressif.com/projects/esp-at/en/latest/esp32/Compile_and_Develop/How_to_clone_project_and_compile_it.html) and the issue is not addressed there.
- [x] I have compiled the hello_world example [Get Started with ESP-IDF](https://docs.espressif.com/projects/esp-at/en/latest/esp32/Compile_and_Develop/How_to_clone_project_and_compile_it.html#step-1-get-started-with-esp-idf) and the issue is not addressed there.
- [x] I have updated my ESP-AT branch (master or release) to the latest version and checked that the issue is present there.
- [x] I have searched the issue tracker for a similar issue and not found a similar issue.

### ESP-AT version

v2.3.0.0_esp8266

### Operating System used

Windows

### If you are using Windows, please specify command line type.

None

### What is the expected behavior?

I expected a response to "AT\r\n" command that I transmit from ArduinoIDE (for simplicity)

### What is the actual behavior?

All the board does is one-time transmit after reset:
`14:20:22.003 ->  ets Jan  8 2013,rst cause:2, boot mode:(3,7)
14:20:22.003 -> 
14:20:22.003 -> load 0x40100000, len 7464, room 16 
14:20:22.052 -> tail 8
14:20:22.052 -> chksum 0x69
14:20:22.052 -> load 0x3ffe8008, len 24, room 0 
14:20:22.052 -> tail 8
14:20:22.052 -> chksum 0xe0
14:20:22.052 -> load 0x3ffe8020, len 3340, room 0 
14:20:22.052 -> tail 12
14:20:22.052 -> chksum 0xc3
14:20:22.052 -> csum 0xc3
14:20:22.291 -> `

### Steps to reproduce

1. Download https://github.com/espressif/esp-at/releases/tag/v2.3.0.0_esp8266
2. Connect the Wemos D1 mini board and locate it's COM port
3. In config file change flash memory to 4MB (originally 2MB) since this is Wemos D1. I flashing without this option, but result is the same
4. Using CHIP DEBUG TOOL upload the firmware

### Build or installation Logs

```plain
Attached installation logs
```

### More Information

Flashing seems to be working with no errors. However I am not able to send an AT command and get a response. I used original config file that is inside release archive. Also I tried modifying 2MB to 4MB. Result is the same. 

[image]
[image]

[image]
[image]

ustccw, 2025-09-05:
@5iemen5 The firmware download should be okay. Could you please try to connect the `RX` of USB2TTL to `D4` of `Wemos D1 mini`, then see what outputs on `D4`.  (baud rate is 115200)
And could you take a picture or video of your hardware connection and operations.

5iemen5, 2025-09-07:
@ustccw you were totally right about GPIO15 (on Wemos D1 mini pin D8, TX) and GPIO13 (on Wemos D1 mini pin D7, RX). It's working fine now

One can easily miss the section of hardware connection for esp8266. [This page,](https://docs.espressif.com/projects/esp-at/en/release-v2.3.0.0_esp8266/Get_Started/index.html) from which one can look for hardware connection is not obviously accessible from [this page](https://docs.espressif.com/projects/esp-at/en/latest/esp32/). Requires to change to 8266 model in the dropdown menu
Also on `readme` file it would be much more convenient for the users to be notified that 8266 is on another branch and is supported. On esp8266 branch, I think, hardware connection should be mentioned as well, in the readme.

ustccw, 2025-09-08:
@5iemen5 Good to hear it‘s working now. When you download the released firmware, the corresponding ESP8266 AT documentation is included in the compressed package. Since the ESP8266 has been replaced by the ESP32-C2, we will gradually remove ESP8266-related resources from our main repository and the AT home page.

closed 2025-11-13 by ustccw